sepfy/libpeer #1: Error:1420B114:SSL routines:ssl_read_internal:uninitialized

ocampana, 2021-04-08:
I just followed you instrictions, after starting the session this is what I get on the console

```
[INFO /home/ottavio/Progetti/pear/src/ice_agent.c:148] SIGNAL: state changed 1 1 failed[5]
[INFO /home/ottavio/Progetti/pear/src/ice_agent.c:148] SIGNAL: state changed 1 1 connecting[2]
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/ottavio/Progetti/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
```

I am on Debian bullseye 64 bots.

tomlankhorst, 2021-04-08:
Same on Ubuntu 20.04 LTS

SA7BNT, 2021-04-08:
Same as @ocampana 
Running Jetson Nano 4gb aarch64 Ubuntu 

> [INFO /home/sa7bnt/pear/src/ice_agent.c:149] SIGNAL: state changed 1 1 failed[5]
[INFO /home/sa7bnt/pear/src/ice_agent.c:149] SIGNAL: state changed 1 1 connecting[2]
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/sa7bnt/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized

regularfry, 2021-04-08:
Same here, rpi zero w running Raspbian buster.

regularfry, 2021-04-08:
Had a quick poke at the code.  [`SSL_read`](https://www.openssl.org/docs/man1.1.0/man3/SSL_read.html) needs [`SSL_set_accept_state`](https://www.openssl.org/docs/man1.1.0/man3/SSL_set_accept_state.html) to have been called first, but the `SSL_set_accept_state` [call](https://github.com/sepfy/pear/blob/main/src/dtls_transport.c#L158) is only reached via a [`g_signal_connect`](https://github.com/sepfy/pear/blob/main/src/ice_agent.c#L186) [callback](https://github.com/sepfy/pear/blob/main/src/ice_agent.c#L57).  I'm not sure what triggers those events, but it sounds async to me.  If that callback doesn't get called before the client connects (or... possibly... ever) then `SSL_read` won't know the connection's in server mode.

sepfy, 2021-04-09:
Thank for your test. I developed this library with Chrome, so I think Chrome is working. When you click start session in the browser, it should callback to [```cb_new_selected_pair_full```](https://github.com/sepfy/pear/blob/main/src/ice_agent.c#L52) and do DTLS handshake. I am checking Firefox in this [branch](https://github.com/sepfy/pear/tree/issue). By increasing idle time, it can pass the handshake and I saw Firefox has received RTP packet , but Firefox still is no video. I will continue to check that. Maybe you can try this with Chrome first.

binhlv2607, 2021-04-09:
Im running in google Chrome but still error
INFO /home/pi/workspaces/pear/src/ice_agent.c:149] SIGNAL: state changed 1 1 failed[5]
[INFO /home/pi/workspaces/pear/src/ice_agent.c:149] SIGNAL: state changed 1 1 connecting[2]
[ERROR /home/pi/workspaces/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/pi/workspaces/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/pi/workspaces/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized
[ERROR /home/pi/workspaces/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized

SA7BNT, 2021-04-09:
> Im running in google Chrome but still error
> INFO /home/pi/workspaces/pear/src/ice_agent.c:149] SIGNAL: state changed 1 1 failed[5]
> [INFO /home/pi/workspaces/pear/src/ice_agent.c:149] SIGNAL: state [...]

Can confirm that on Chromium aarch64 Ubuntu

regularfry, 2021-04-09:
Yep. Chrome here too.  Possibly relevant: I'm testing over the network, not locally: `index.html` is served up by a `python -m SimpleHTTPServer`.  I don't *think* that should matter, if I've understood what libnice is doing correctly.

SA7BNT, 2021-04-09:
> Yep. Chrome here too. Possibly relevant: I'm testing over the network, not locally. `index.html` is served up by a `python -m SimpleHTTPServer`. I don't _think_ that should matter, if I've understood what libnice is doing correctly.

Running on index.html on localhost.

binhlv2607, 2021-04-09:
> Can confirm that on Chromium aarch64 Ubuntu

yep. I try to running index.html by google chrome in ubuntu ( my PC) and in my Raspberry, still error. How can i by pass SSL to tranfer video success?

ocampana, 2021-04-09:
> 
> 
> Yep. Chrome here too. Possibly relevant: I'm testing over the network, not locally. `index.html` is served up by a `python -m SimpleHTTPServer`. I don't _think_ that should matter, if I've understood what libnice is doing correctly.

As far as I know, WebRTC does not work over HTTP, it mandates HTTPS

sepfy, 2021-04-09:
It's weird. My google chrome version is 85.0.4183.83 (Linux Mint 20)
Maybe try this [branch](https://github.com/sepfy/pear/tree/issue)
Or remove the lines [229](https://github.com/sepfy/pear/blob/issue/src/ice_agent.c#L229) and [230](https://github.com/sepfy/pear/blob/issue/src/ice_agent.c#L230) in ice_agent.c

I think that serve by SimpleHTTPServer or locally both are ok.

binhlv2607, 2021-04-09:
> It's weird. My google chrome version is 85.0.4183.83 (Linux Mint 20)
> Maybe try this [branch](https://github.com/sepfy/pear/tree/issue)
> Or remove the lines [...]

im try in new branch and remove 2 lines of your. still error.

sepfy, 2021-04-09:
could you provide your remote sdp ?

binhlv2607, 2021-04-09:
> could you provide your remote sdp ?

here:
dj0wDQpvPS0gNjE2MjA4NjE5ODYwNjgwMjUzNiAyIElOIElQNCAxMjcuMC4wLjENCnM9LQ0KdD0wIDANCmE9Z3JvdXA6QlVORExFIDANCmE9ZXh0bWFwLWFsbG93LW1peGVkDQphPW1zaWQtc2VtYW50aWM6IFdNUw0KbT12aWRlbyAzNjgxNCBVRFAvVExTL1JUUC9TQVZQRiA5NiA5NyA5OCA5OSAxMDAgMTAxIDEwMiAxMjEgMTI3IDEyMCAxMjUgMTA3IDEwOCAxMDkgMTI0IDExOSAxMjMNCmM9SU4gSVA0IDE0LjIzMi4xNjQuMTAxDQphPXJ0Y3A6OSBJTiBJUDQgMC4wLjAuMA0KYT1jYW5kaWRhdGU6ODQyMTYzMDQ5IDEgdWRwIDE2Nzc3Mjk1MzUgMTQuMjMyLjE2NC4xMDEgMzY4MTQgdHlwIHNyZmx4IHJhZGRyIDAuMC4wLjAgcnBvcnQgMCBnZW5lcmF0aW9uIDAgbmV0d29yay1jb3N0IDk5OQ0KYT1pY2UtdWZyYWc6a09XQw0KYT1pY2UtcHdkOnZYQTNTYkVtMGdyeHBtTmVXelByaHdTbg0KYT1pY2Utb3B0aW9uczp0cmlja2xlDQphPWZpbmdlcnByaW50OnNoYS0yNTYgM0Y6RDg6RjA6ODk6RDQ6RjU6RDQ6NzU6MDk6QkE6OEM6MDM6N0M6NjM6NTk6QjU6RjU6RjY6QTk6MzE6Qjg6OTA6QzU6Nzc6REY6OEE6Njg6NUU6RTg6MkY6M0I6MUINCmE9c2V0dXA6YWN0cGFzcw0KYT1taWQ6MA0KYT1leHRtYXA6MSB1cm46aWV0ZjpwYXJhbXM6cnRwLWhkcmV4dDp0b2Zmc2V0DQphPWV4dG1hcDoyIGh0dHA6Ly93d3cud2VicnRjLm9yZy9leHBlcmltZW50cy9ydHAtaGRyZXh0L2Ficy1zZW5kLXRpbWUNCmE9ZXh0bWFwOjMgdXJuOjNncHA6dmlkZW8tb3JpZW50YXRpb24NCmE9ZXh0bWFwOjQgaHR0cDovL3d3dy5pZXRmLm9yZy9pZC9kcmFmdC1ob2xtZXItcm1jYXQtdHJhbnNwb3J0LXdpZGUtY2MtZXh0ZW5zaW9ucy0wMQ0KYT1leHRtYXA6NSBodHRwOi8vd3d3LndlYnJ0Yy5vcmcvZXhwZXJpbWVudHMvcnRwLWhkcmV4dC9wbGF5b3V0LWRlbGF5DQphPWV4dG1hcDo2IGh0dHA6Ly93d3cud2VicnRjLm9yZy9leHBlcmltZW50cy9ydHAtaGRyZXh0L3ZpZGVvLWNvbnRlbnQtdHlwZQ0KYT1leHRtYXA6NyBodHRwOi8vd3d3LndlYnJ0Yy5vcmcvZXhwZXJpbWVudHMvcnRwLWhkcmV4dC92aWRlby10aW1pbmcNCmE9ZXh0bWFwOjggaHR0cDovL3d3dy53ZWJydGMub3JnL2V4cGVyaW1lbnRzL3J0cC1oZHJleHQvY29sb3Itc3BhY2UNCmE9ZXh0bWFwOjkgdXJuOmlldGY6cGFyYW1zOnJ0cC1oZHJleHQ6c2RlczptaWQNCmE9ZXh0bWFwOjEwIHVybjppZXRmOnBhcmFtczpydHAtaGRyZXh0OnNkZXM6cnRwLXN0cmVhbS1pZA0KYT1leHRtYXA6MTEgdXJuOmlldGY6cGFyYW1zOnJ0cC1oZHJleHQ6c2RlczpyZXBhaXJlZC1ydHAtc3RyZWFtLWlkDQphPXNlbmRyZWN2DQphPW1zaWQ6LSA0NWU4YTFlYi0xY2MzLTQyYjctYmFmNC01ZGY4NmFlMjJiYzUNCmE9cnRjcC1tdXgNCmE9cnRjcC1yc2l6ZQ0KYT1ydHBtYXA6OTYgVlA4LzkwMDAwDQphPXJ0Y3AtZmI6OTYgZ29vZy1yZW1iDQphPXJ0Y3AtZmI6OTYgdHJhbnNwb3J0LWNjDQphPXJ0Y3AtZmI6OTYgY2NtIGZpcg0KYT1ydGNwLWZiOjk2IG5hY2sNCmE9cnRjcC1mYjo5NiBuYWNrIHBsaQ0KYT1ydHBtYXA6OTcgcnR4LzkwMDAwDQphPWZtdHA6OTcgYXB0PTk2DQphPXJ0cG1hcDo5OCBWUDkvOTAwMDANCmE9cnRjcC1mYjo5OCBnb29nLXJlbWINCmE9cnRjcC1mYjo5OCB0cmFuc3BvcnQtY2MNCmE9cnRjcC1mYjo5OCBjY20gZmlyDQphPXJ0Y3AtZmI6OTggbmFjaw0KYT1ydGNwLWZiOjk4IG5hY2sgcGxpDQphPWZtdHA6OTggcHJvZmlsZS1pZD0wDQphPXJ0cG1hcDo5OSBydHgvOTAwMDANCmE9Zm10cDo5OSBhcHQ9OTgNCmE9cnRwbWFwOjEwMCBWUDkvOTAwMDANCmE9cnRjcC1mYjoxMDAgZ29vZy1yZW1iDQphPXJ0Y3AtZmI6MTAwIHRyYW5zcG9ydC1jYw0KYT1ydGNwLWZiOjEwMCBjY20gZmlyDQphPXJ0Y3AtZmI6MTAwIG5hY2sNCmE9cnRjcC1mYjoxMDAgbmFjayBwbGkNCmE9Zm10cDoxMDAgcHJvZmlsZS1pZD0yDQphPXJ0cG1hcDoxMDEgcnR4LzkwMDAwDQphPWZtdHA6MTAxIGFwdD0xMDANCmE9cnRwbWFwOjEwMiBIMjY0LzkwMDAwDQphPXJ0Y3AtZmI6MTAyIGdvb2ctcmVtYg0KYT1ydGNwLWZiOjEwMiB0cmFuc3BvcnQtY2MNCmE9cnRjcC1mYjoxMDIgY2NtIGZpcg0KYT1ydGNwLWZiOjEwMiBuYWNrDQphPXJ0Y3AtZmI6MTAyIG5hY2sgcGxpDQphPWZtdHA6MTAyIGxldmVsLWFzeW1tZXRyeS1hbGxvd2VkPTE7cGFja2V0aXphdGlvbi1tb2RlPTE7cHJvZmlsZS1sZXZlbC1pZD00MjAwMWYNCmE9cnRwbWFwOjEyMSBydHgvOTAwMDANCmE9Zm10cDoxMjEgYXB0PTEwMg0KYT1ydHBtYXA6MTI3IEgyNjQvOTAwMDANCmE9cnRjcC1mYjoxMjcgZ29vZy1yZW1iDQphPXJ0Y3AtZmI6MTI3IHRyYW5zcG9ydC1jYw0KYT1ydGNwLWZiOjEyNyBjY20gZmlyDQphPXJ0Y3AtZmI6MTI3IG5hY2sNCmE9cnRjcC1mYjoxMjcgbmFjayBwbGkNCmE9Zm10cDoxMjcgbGV2ZWwtYXN5bW1ldHJ5LWFsbG93ZWQ9MTtwYWNrZXRpemF0aW9uLW1vZGU9MDtwcm9maWxlLWxldmVsLWlkPTQyMDAxZg0KYT1ydHBtYXA6MTIwIHJ0eC85MDAwMA0KYT1mbXRwOjEyMCBhcHQ9MTI3DQphPXJ0cG1hcDoxMjUgSDI2NC85MDAwMA0KYT1ydGNwLWZiOjEyNSBnb29nLXJlbWINCmE9cnRjcC1mYjoxMjUgdHJhbnNwb3J0LWNjDQphPXJ0Y3AtZmI6MTI1IGNjbSBmaXINCmE9cnRjcC1mYjoxMjUgbmFjaw0KYT1ydGNwLWZiOjEyNSBuYWNrIHBsaQ0KYT1mbXRwOjEyNSBsZXZlbC1hc3ltbWV0cnktYWxsb3dlZD0xO3BhY2tldGl6YXRpb24tbW9kZT0xO3Byb2ZpbGUtbGV2ZWwtaWQ9NDJlMDFmDQphPXJ0cG1hcDoxMDcgcnR4LzkwMDAwDQphPWZtdHA6MTA3IGFwdD0xMjUNCmE9cnRwbWFwOjEwOCBIMjY0LzkwMDAwDQphPXJ0Y3AtZmI6MTA4IGdvb2ctcmVtYg0KYT1ydGNwLWZiOjEwOCB0cmFuc3BvcnQtY2MNCmE9cnRjcC1mYjoxMDggY2NtIGZpcg0KYT1ydGNwLWZiOjEwOCBuYWNrDQphPXJ0Y3AtZmI6MTA4IG5hY2sgcGxpDQphPWZtdHA6MTA4IGxldmVsLWFzeW1tZXRyeS1hbGxvd2VkPTE7cGFja2V0aXphdGlvbi1tb2RlPTA7cHJvZmlsZS1sZXZlbC1pZD00MmUwMWYNCmE9cnRwbWFwOjEwOSBydHgvOTAwMDANCmE9Zm10cDoxMDkgYXB0PTEwOA0KYT1ydHBtYXA6MTI0IHJlZC85MDAwMA0KYT1ydHBtYXA6MTE5IHJ0eC85MDAwMA0KYT1mbXRwOjExOSBhcHQ9MTI0DQphPXJ0cG1hcDoxMjMgdWxwZmVjLzkwMDAwDQphPXNzcmMtZ3JvdXA6RklEIDM1ODE5OTM4OTEgMjAyOTg5MzkxDQphPXNzcmM6MzU4MTk5Mzg5MSBjbmFtZTpQYUx0Uk1HeFBvOVUyV3lhDQphPXNzcmM6MzU4MTk5Mzg5MSBtc2lkOi0gNDVlOGExZWItMWNjMy00MmI3LWJhZjQtNWRmODZhZTIyYmM1DQphPXNzcmM6MzU4MTk5Mzg5MSBtc2xhYmVsOi0NCmE9c3NyYzozNTgxOTkzODkxIGxhYmVsOjQ1ZThhMWViLTFjYzMtNDJiNy1iYWY0LTVkZjg2YWUyMmJjNQ0KYT1zc3JjOjIwMjk4OTM5MSBjbmFtZTpQYUx0Uk1HeFBvOVUyV3lhDQphPXNzcmM6MjAyOTg5MzkxIG1zaWQ6LSA0NWU4YTFlYi0xY2MzLTQyYjctYmFmNC01ZGY4NmFlMjJiYzUNCmE9c3NyYzoyMDI5ODkzOTEgbXNsYWJlbDotDQphPXNzcmM6MjAyOTg5MzkxIGxhYmVsOjQ1ZThhMWViLTFjYzMtNDJiNy1iYWY0LTVkZjg2YWUyMmJjNQ0K

GrubbyHalo, 2021-04-13:
I could not get this to work. I am getting the same error....
`[ERROR /home/pi/pear/src/dtls_transport.c:185] error:1420B114:SSL routines:ssl_read_internal:uninitialized`
I am using Brave browser (Based on chrome) and the latest raspberry pi OS.

ikamthania, 2021-04-16:
Same error using Ubuntu and chrome

sepfy, 2021-04-19:
Hi all,

I spent some time to test different environment. I ensure that the following environment are ok:
1. Ubuntu 20.04 with chromium (install by apt-get chromium-browser).
2. Raspberry Pi OS on Pi 4 with default chromium
3. Jetson nano official [SD card image](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write) with default chromium

If you use these environment, but still has some issues. I guess that reasons are network or instructions of example. For the instruction issue, maybe my document is not clear. So I created another [example](https://github.com/sepfy/pear/tree/gstreamer/examples/signal). It can help to exchange SDP by HTTP server.  You can test it in the [branch](https://github.com/sepfy/pear/tree/gstreamer).Thanks

0xsky, 2021-06-10:
I got the same error msg
    **error:140DF114:SSL routines:SSL_read:uninitialized**
 I found that the main reason was that the cb_new_selected_pair_full callback was not called，
But the cause hasn't been found yet

sepfy, 2023-08-14:
switch to mbedtls

closed 2023-08-14 by sepfy