sepfy/libpeer #3: compile issue

lqian, 2021-06-29:
Hi, 
Pear is a interesting project. I try to compile the project for a tour with Ubuntu16.0.4, python3.6, etc. but get link error that seems it is not compatible with OpenSSL.  also It link error with Openssl-1.1.1g.
Is there any compile requirements for the project.
<pre>
[ 75%] Linking C executable local_file
../../src/libpear.so: undefined reference to `BIO_set_data'
../../src/libpear.so: undefined reference to `BIO_set_data'
../../src/libpear.so: undefined reference to `BIO_set_init'
../../src/libpear.so: undefined reference to `BIO_meth_set_destroy'
../../src/libpear.so: undefined reference to `BIO_meth_set_write'
/train-data/pear/third_party/libnice/builddir/nice/libnice.so: undefined reference to `g_ptr_array_find'
../../src/libpear.so: undefined reference to `BIO_get_data'
../../src/libpear.so: undefined reference to `BIO_set_shutdown'
../../src/libpear.so: undefined reference to `BIO_meth_new'
../../src/libpear.so: undefined reference to `BIO_meth_set_create'
../../src/libpear.so: undefined reference to `BIO_meth_set_ctrl'
</pre>

sepfy, 2021-07-01:
Hi

I tried to build Pear in Ubuntu 16.04 (Docker). I also met your issue.
The solution is that upgrade openssl(1.1.1) and glib(2.58)

```
libnice.so: undefined reference to `g_ptr_array_find' --> glib
undefined reference to `BIO_set_data' --> ssl
```

But I saw that you already upgrade openssl to 1.1.1g, can you help me to confirm that is the linked library correct? Because the installation path of openssl which is built by myself is /usr/local

lqian, 2021-07-10:
it seems the default library path makes a compile trouble.  manually add '/usr/local/lib' and '/usr/local/lib/x86_64-linux-gnu' library path to compile command work good.

```
 /usr/bin/cc -rdynamic CMakeFiles/local_file.dir/h264_packetizer.c.o CMakeFiles/local_file.dir/h264_parser.c.o CMakeFiles/local_file.dir/main.c.o -o local_file -L/usr/local/lib -L/usr/local/lib/x86_64-linux-gnu  -L/train-data/pear/third_party/libnice/builddir/nice  -L/train-data/pear/third_party/libsrtp/build  -L/train-data/pear/third_party/media-server/librtp/debug.linux  -Wl,-rpath,/train-data/pear/third_party/libnice/builddir/nice:/train-data/pear/third_party/libsrtp/build:/train-data/pear/third_party/media-server/librtp/debug.linux:/train-data/pear/build/src ../../src/libpear.so -lglib-2.0 -lssl -lcrypto -lnice -lsrtp2 -lrtp -lpthread

```

sepfy, 2023-08-14:
Changed to use mbedtls

closed 2023-08-14 by sepfy