sepfy/libpeer #5: run local_file issues

lbit20131014, 2021-09-08:
I don't know how to use the local_file example, and after running it, it prints the following：

eyJ0eXBlIjogImFuc3dlciIsICJzZHAiOiAidj0wXHJcbm89LSAxNDk1Nzk5ODExMDg0OTcwIDE0OTU3OTk4MTEwODQ5NzAgSU4gSVA0IDAuMC4wLjBcclxucz1TdHJlYW1pbmcgVGVzdFxyXG50PTAgMFxyXG5hPWdyb3VwOkJVTkRMRSAwXHJcbmE9bXNpZC1zZW1hbnRpYzogcGVhclxyXG5tPXZpZGVvIDkgVURQL1RMUy9SVFAvU0FWUEYgMTAyXHJcbmM9SU4gSVA0IDAuMC4wLjBcclxuYT1zZW5kb25seVxyXG5hPW1pZDowXHJcbmE9cnRjcC1tdXhcclxuYT1pY2UtdWZyYWc6UXlGMVxyXG5hPWljZS1wd2Q6R3FVTHhHdTBLWU4yNGZtQk4xUVFEZ1xyXG5hPWljZS1vcHRpb25zOnRyaWNrbGVcclxuYT1maW5nZXJwcmludDpzaGEtMjU2IEE0OkU2Ojg4OkYzOjlBOkI2OjBFOkRFOkQ2OjM5Ojc5OkQxOjBEOjY5Ojc3OjAwOjc5OkE1OjE0OkFCOjczOkNEOkU3OjdCOjc0OkU4OkU2OjJDOjMzOkYwOjhEOjQ5XHJcbmE9c2V0dXA6cGFzc2l2ZVxyXG5hPXJ0cG1hcDoxMDIgSDI2NC85MDAwMFxyXG5hPWZtdHA6MTAyIHBhY2tldGl6YXRpb24tbW9kZT0xXHJcbmE9cnRjcC1mYjoxMDIgbmFja1xyXG5hPXJ0Y3AtZmI6MTAyIG5hY2sgcGxpXHJcbmE9cnRjcC1mYjoxMDIgZ29vZy1yZW1iXHJcbmE9Y2FuZGlkYXRlOjEgMSB1ZHAgMjAxNTM2MzMyNyAxMC4xMS4xNy4xNjcgMzM4MjAgdHlwIGhvc3RcclxuYT1jYW5kaWRhdGU6MiAxIHVkcCAxMDE1MDIxODIzIDEwLjExLjE3LjE2NyAwIHR5cCBob3N0XHJcbmE9Y2FuZGlkYXRlOjMgMSB1ZHAgMTAxMDgyNzUxOSAxMC4xMS4xNy4xNjcgNDIyMTUgdHlwIGhvc3RcclxuYT1jYW5kaWRhdGU6NyAxIHVkcCAxNjc5ODE5MDA3IDU4LjI1MC4yMy4xOTQgMzM4MjAgdHlwIHNyZmx4XHJcbmE9Y2FuZGlkYXRlOjggMSB1ZHAgODQ3MjQ5NjYzIDU4LjI1MC4yMy4xOTQgMCB0eXAgc3JmbHhcclxuYT1jYW5kaWRhdGU6OSAxIHVkcCA4NDMwNTUzNTkgNTguMjUwLjIzLjE5NCA0MjIxNSB0eXAgc3JmbHhcclxuIn0=

** (process:18272): WARNING **: 19:56:49.715: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 5 seconds

** (process:18272): WARNING **: 19:56:49.715: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 5 seconds

** (process:18272): WARNING **: 19:57:00.295: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 10 seconds

** (process:18272): WARNING **: 19:57:00.295: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 10 seconds

** (process:18272): WARNING **: 19:57:15.297: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 20 seconds

** (process:18272): WARNING **: 19:57:15.297: Failed to GET http://192.168.5.1:5500/rootDesc.xml: Could not connect: Socket I/O timed out, retrying in 20 seconds

** (process:18272): WARNING **: 19:57:40.297: Maximum number of retries failed, not trying again

** (process:18272): WARNING **: 19:57:40.297: Maximum number of retries failed, not trying again

lbit20131014, 2021-09-08:
From which module is this warning print output

sepfy, 2021-11-25:
If there is no problem, I will close this issue. Thanks.

closed 2022-01-02 by sepfy