sepfy/libpeer #8: recording example receives/records nothing

AleXoundOS, 2021-11-24:
Browser asks to select screencast capture source, I choose one. In console it prints just:
```
Listening 0.0.0.0:8000
```
No file gets created in the directory of executable.
Tested both with chrome and chromium with the same result.

sepfy, 2021-11-25:
Do you connect with internet? Because the default STUN server is 18.191.223.12

AleXoundOS, 2021-11-25:
Yes.
I tracked down the problem and here is the workaround diff which makes example work (`record.h264` gets saved):
```diff
--- a/examples/recording/index_html.h
+++ b/examples/recording/index_html.h
@@ -13,6 +13,7 @@ const char index_html[] = " \
   <body> \n \
     <video style='display:block; margin: 0 auto;' id='localVideo' autoplay muted></video> \n \
     <script> \n \
+      var counter=0; \n \
       function jsonRpc(payload, cb) { \n \
         var xhttp = new XMLHttpRequest(); \n \
         xhttp.onreadystatechange = function() { \n \
@@ -45,7 +46,14 @@ console.log(JSON.parse(atob(sdp))); \n\
         } \n \
       } \n \
       pc.onicecandidate = event => { \n \
-        if (event.candidate === null) { \n \
+        counter++; \n \
+        console.log(\"counter = \" + String(counter)); \n \
+        console.log(event); \n \
+        if (event.candidate === null || event.candidate.address.length > 15) { \n \
+          console.log(\"ignoring this event\") \n \
+        } \n \
+        else \n \
+        { \n \
           let sdp = pc.localDescription.sdp; \n \
           var payload = {\"jsonrpc\": \"2.0\", \"method\": \"call\", \"params\": btoa(sdp)}; \n \
           jsonRpc(payload, jsonRpcHandle); \n \
```

(`counter` is used only for debugging.)

sepfy, 2021-11-25:
OK, Thanks!

AleXoundOS, 2021-11-25:
I have no experience with javascript and the workaround was found just empirically. It definitely needs a more correct approach.

sepfy, 2021-11-26:
I see you ignoring the ice candidate which the length of address is larger than 15. I guess it is a mDNS address. I tried to [remove it and transfer to a ipv4 address](https://github.com/sepfy/pear/blob/8293ca0cbc673726d28def2e0610ac1ca9198c5c/src/peer_connection.c#L334), but it still seems to have some problems. I will check that.

AleXoundOS, 2021-11-26:
I'm not only ignoring candidates which are not IPs basically, but since there is only one IP candidate, **`jsonRpc` is called only once**. Otherwise, `setRemoteDescription` was called 2 times what caused an error.
Without my changes `jsonRpc` was not called at all, because the condition for calling was `event.candidate === null`.

AleXoundOS, 2021-11-26:
I used `chrome://webrtc-internals` for debugging.

sepfy, 2021-11-29:
According to the [onicecanidate API](https://developer.mozilla.org/en-US/docs/Web/API/RTCPeerConnection/onicecandidate), if the event's candidate property is null, ICE gathering has finished. It should be waited until all candidates are obtained, then send the offer. So you don't get event with null candidate property?

AleXoundOS, 2021-11-29:
> So you don't get event with null candidate property?

It seems not. Log without workaround:

```
11/29/2021, 2:07:23 PM	transceiverAdded
11/29/2021, 2:07:23 PM	createOffer
11/29/2021, 2:07:23 PM	negotiationneeded
11/29/2021, 2:07:23 PM	createOfferOnSuccess (type: "offer", 2 sections)
11/29/2021, 2:07:23 PM	setLocalDescription (type: "offer", 2 sections)
11/29/2021, 2:07:23 PM	setLocalDescriptionOnSuccess
11/29/2021, 2:07:23 PM	signalingstatechange
11/29/2021, 2:07:23 PM	transceiverModified
11/29/2021, 2:07:23 PM	icegatheringstatechange
11/29/2021, 2:07:23 PM	icecandidate (host)
11/29/2021, 2:07:23 PM	icecandidate (srflx)
```

sepfy, 2022-01-02:
After getting null candidate, the browser should call `jsonRpc` function to send the offer to recording program. When recording program accepts the offer and response an answer, browser should be callback to `jsonRpcHandle` and `setRemoteDescription`.
I think the recording program doesn't response the answer, because mDNS parse failed or it need more time to resolve.

I created a  [branch](https://github.com/sepfy/pear/tree/issue-8) to disable mDNS resolver. If possible, please try again. Thanks.

sepfy, 2023-10-22:
This example is no longer supported

closed 2023-10-22 by sepfy