sepfy/libpeer #11: Not showing any thing & missing singal_service.h

absun-pk, 2022-03-19:
I have tried all examples on rpi4 with 8GB page loads but shows nothing, whereas the GStreamer example is missing "signal_service.h".

sepfy, 2022-03-22:
The GStreamer example is deprecated, I will remove it. Does web page has any content? Can you try the video_on_demand example first? Make sure that the test.264 file provided by the README is in the same directory.

sepfy, 2022-04-01:
Hi aldoshkind 

There is no technical problems in integrating with gstreamer. Actually, almost examples are based on gstreamer. I just classify examples by scenario. The gstreamer examples has been renamed to [surveillance](https://github.com/sepfy/pear/tree/main/examples/surveillance) and is suite for Raspberry Pi. You can modify the gstreamer pipeline to anything you want, like changing encoder from omxh264 to libx264
```
video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! queue ! x264enc bitrate=6000 speed-preset=ultrafast tune=zerolatency key-int-max=15 ! video/x-h264,profile=constrained-baseline
```

aldoshkind, 2022-04-01:
@sepfy thank you alot!
I managed to revive `gstreamer` example and it works by now. Comparing `gstreamer` example against `surveillance` I found interesting difference: in `surveillance` you do
```c
  static int pt = -1;
  // Update payload type of rtph264pay
  int gst_pt = gst_pt = peer_connection_get_rtpmap(g_surveillance.pc, CODEC_H264);

  if(pt != gst_pt) {
    pt = gst_pt;
    g_object_set(g_surveillance.rtp, "pt", pt, NULL);
  }
```
Can you please explain why you do this?

sepfy, 2022-04-04:
Hi aldoshkind

The payload type of the codec defined in the SDP generated by each browser will be different, so we need to set the payload type (pt) of rtph264pay to the value defined in the SDP. When the browser receives the RTP packet, it knows which codec of packet is.

For example, If you print the SDP of offer, there are some information as below:
```
a=rtpmap:96 VP8/90000
...
a=rtpmap:98 VP9/90000
...
a=rtpmap:102 H264/90000
```

sepfy, 2023-10-22:
This example is no longer supported

closed 2023-10-22 by sepfy