sepfy/libpeer #13: STUN server address and working without STUN at all

aldoshkind, 2022-04-01:
Hello @sepfy !
In `peer_connection.c` I see following:
```c
static const gchar *STUN_ADDR = "18.191.223.12";
static const guint STUN_PORT = 3478;
```
What is this server?
It takes approx 20 seconds to establish connection when I use these settings as defaults. If I use `stun.l.google.com:19302` instead connection gets established in a second.

And what about working in local isolated network? How to set everything up correctly? What STUN server settings shoul I use? Is it possible to avoid using STUN server?

sepfy, 2022-04-04:
This server is provided by libnice. Of course you can change to google stun server.

I am not sure whether it is possible to avoid using stun server or not, but if you want to use local isolated network, you can consider run a stun server on your machine. The stun server has been built in ```pear/third_party/libnice/builddir/stun/tools```.
Then modify ```STUN_ADDR``` in ```pear/src/peer_connection.c``` and ```iceServers``` of ```pear/examples/surveillance/index.html``` to your local IP address.

And if you want to use google stun server, you need to try to build the Google [WebRTC native code](https://webrtc.github.io/webrtc-org/native-code/). It will generate a stun server binary.

aldoshkind, 2022-04-05:
My question was not exact and clear enough as I am newbie to WebRTC.
Problem was that examples was not working if I set `iceServers = []` in index.html
After some digging I found that enabling mDNS solves the problem.
But while it works in Chromium/Opera, it doesn't in Firefox, possibly due to this bug: [https://bugzilla.mozilla.org/show_bug.cgi?id=1691189](bug). I use Ubuntu 18.04 and 20.04 and browser binaries from official ubuntu repos.
Have you had any experience using pear with Firefox?

sepfy, 2022-04-06:
I have tried using Firefox, but it doesn't work.
If there is a candidate in SDP is ```xxxx.local```, it means you need to enable mDNS. When I tested with Firefox, I checked there is no candidate using mDNS, so I guess it has nothing to do with mDNS. I'm still debugging

You can check your offer SDP with browser console by modifying code as bellow
```
diff --git a/examples/surveillance/index_html.h b/examples/surveillance/index_html.h
index 00a194f..db706e4 100644
--- a/examples/surveillance/index_html.h
+++ b/examples/surveillance/index_html.h
@@ -18,6 +18,7 @@ const char index_html[] = " \
       }); \n \
       var log = msg => { console.log(msg); }; \n \
       function sendOfferToCall(sdp) { \n \
+        console.log(sdp);\n \
         var xhttp = new XMLHttpRequest(); \n \
         xhttp.onreadystatechange = function() { \n \
           if (this.readyState == 4 && this.status == 200) { \n \
```

aldoshkind, 2022-10-21:
In our project we disable mdns. Firefox indeed has strange behavior with mdns, but the problem may be relaxed by disabling mdns obfuscation (go to `about:config` and turn off `media.peerconnection.ice.obfuscate_host_addresses`)
There was another problems in supporting Firefox, I can share my expirience in separate issue if you are interested. Also you can investigate my fork here https://github.com/aldoshkind/pear

closed 2023-01-21 by aldoshkind