sepfy/libpeer #15: The P2P can be established,but the video not display on chrome browse

robertqd, 2022-04-17:
Hi @sepfy 
could you help me? 
The description as follown:
one termination is chrome(chrome/100.0.4896.127), another termiation is that pear was run at ubuntu on virtual machine. The P2P has been established. we can see the video data can be sent frome pear to chrome from wireshark. but the video doesn't display on chrome.  
**the sdp offer:**
:48	
createOfferOnSuccess (type: "offer", 2 sections)
Copy description to clipboard
v=0 (6 more lines)
o=- 8752117117835402785 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=extmap-allow-mixed
a=msid-semantic: WMS
m=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 127 121 125 107 108 109 124 120 123 119 35 36 41 42 114 115 116 117 118 (133 more lines) mid=0
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=ice-ufrag:jnf4
a=ice-pwd:bd90gvRbdL46FjqTu3AQ/EqY
a=ice-options:trickle
a=fingerprint:sha-256 1D:D5:DB:DB:D7:CA:B8:AA:7F:61:9A:74:0E:04:50:EE:41:5C:21:DB:42:B2:98:3F:4D:4F:91:5B:7B:B8:71:35
a=setup:actpass
a=mid:0
a=extmap:1 urn:ietf:params:rtp-hdrext:toffset
a=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:3 urn:3gpp:video-orientation
a=extmap:4 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01
a=extmap:5 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay
a=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type
a=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing
a=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/color-space
a=extmap:9 urn:ietf:params:rtp-hdrext:sdes:mid
a=extmap:10 urn:ietf:params:rtp-hdrext:sdes:rtp-stream-id
a=extmap:11 urn:ietf:params:rtp-hdrext:sdes:repaired-rtp-stream-id
a=sendrecv
a=msid:- 4c46546f-7651-420e-b1af-8754bd3ea3ad
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:96 VP8/90000
a=rtcp-fb:96 goog-remb
a=rtcp-fb:96 transport-cc
a=rtcp-fb:96 ccm fir
a=rtcp-fb:96 nack
a=rtcp-fb:96 nack pli
a=rtpmap:97 rtx/90000
a=fmtp:97 apt=96
a=rtpmap:98 VP9/90000
a=rtcp-fb:98 goog-remb
a=rtcp-fb:98 transport-cc
a=rtcp-fb:98 ccm fir
a=rtcp-fb:98 nack
a=rtcp-fb:98 nack pli
a=fmtp:98 profile-id=0
a=rtpmap:99 rtx/90000
a=fmtp:99 apt=98
a=rtpmap:100 VP9/90000
a=rtcp-fb:100 goog-remb
a=rtcp-fb:100 transport-cc
a=rtcp-fb:100 ccm fir
a=rtcp-fb:100 nack
a=rtcp-fb:100 nack pli
a=fmtp:100 profile-id=2
a=rtpmap:101 rtx/90000
a=fmtp:101 apt=100
a=rtpmap:127 H264/90000
a=rtcp-fb:127 goog-remb
a=rtcp-fb:127 transport-cc
a=rtcp-fb:127 ccm fir
a=rtcp-fb:127 nack
a=rtcp-fb:127 nack pli
a=fmtp:127 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f
a=rtpmap:121 rtx/90000
a=fmtp:121 apt=127
a=rtpmap:125 H264/90000
a=rtcp-fb:125 goog-remb
a=rtcp-fb:125 transport-cc
a=rtcp-fb:125 ccm fir
a=rtcp-fb:125 nack
a=rtcp-fb:125 nack pli
a=fmtp:125 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42001f
a=rtpmap:107 rtx/90000
a=fmtp:107 apt=125
a=rtpmap:108 H264/90000
a=rtcp-fb:108 goog-remb
a=rtcp-fb:108 transport-cc
a=rtcp-fb:108 ccm fir
a=rtcp-fb:108 nack
a=rtcp-fb:108 nack pli
a=fmtp:108 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f
a=rtpmap:109 rtx/90000
a=fmtp:109 apt=108
a=rtpmap:124 H264/90000
a=rtcp-fb:124 goog-remb
a=rtcp-fb:124 transport-cc
a=rtcp-fb:124 ccm fir
a=rtcp-fb:124 nack
a=rtcp-fb:124 nack pli
a=fmtp:124 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42e01f
a=rtpmap:120 rtx/90000
a=fmtp:120 apt=124
a=rtpmap:123 H264/90000
a=rtcp-fb:123 goog-remb
a=rtcp-fb:123 transport-cc
a=rtcp-fb:123 ccm fir
a=rtcp-fb:123 nack
a=rtcp-fb:123 nack pli
a=fmtp:123 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=4d001f
a=rtpmap:119 rtx/90000
a=fmtp:119 apt=123
a=rtpmap:35 H264/90000
a=rtcp-fb:35 goog-remb
a=rtcp-fb:35 transport-cc
a=rtcp-fb:35 ccm fir
a=rtcp-fb:35 nack
a=rtcp-fb:35 nack pli
a=fmtp:35 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=4d001f
a=rtpmap:36 rtx/90000
a=fmtp:36 apt=35
a=rtpmap:41 AV1/90000
a=rtcp-fb:41 goog-remb
a=rtcp-fb:41 transport-cc
a=rtcp-fb:41 ccm fir
a=rtcp-fb:41 nack
a=rtcp-fb:41 nack pli
a=rtpmap:42 rtx/90000
a=fmtp:42 apt=41
a=rtpmap:114 H264/90000
a=rtcp-fb:114 goog-remb
a=rtcp-fb:114 transport-cc
a=rtcp-fb:114 ccm fir
a=rtcp-fb:114 nack
a=rtcp-fb:114 nack pli
a=fmtp:114 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=64001f
a=rtpmap:115 rtx/90000
a=fmtp:115 apt=114
a=rtpmap:116 red/90000
a=rtpmap:117 rtx/90000
a=fmtp:117 apt=116
a=rtpmap:118 ulpfec/90000
a=ssrc-group:FID 2322705154 2857601720
a=ssrc:2322705154 cname:ZmMkzc9v9XyhMLOi
a=ssrc:2322705154 msid:- 4c46546f-7651-420e-b1af-8754bd3ea3ad
a=ssrc:2322705154 mslabel:-
a=ssrc:2322705154 label:4c46546f-7651-420e-b1af-8754bd3ea3ad
a=ssrc:2857601720 cname:ZmMkzc9v9XyhMLOi
a=ssrc:2857601720 msid:- 4c46546f-7651-420e-b1af-8754bd3ea3ad
a=ssrc:2857601720 mslabel:-
a=ssrc:2857601720 label:4c46546f-7651-420e-b1af-8754bd3ea3ad

the sdp answer：
setRemoteDescription (type: "answer", 2 sections)
Copy description to clipboard
v=0 (5 more lines)
o=- 1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: WMS
a=group:BUNDLE 0
m=video 9 UDP/TLS/RTP/SAVPF 127 121 (24 more lines) mid=0
a=rtpmap:127 H264/90000
a=rtcp-fb:127 goog-remb
a=rtcp-fb:127 transport-cc
a=rtcp-fb:127 com fir
a=rtcp-fb:127 nack
a=rtcp-fb:127 nack pli
a=fmtp:127 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f
a=rtpmap:121 rtx/90000
a=fmtp:121 apt=127
c=IN IP4 0.0.0.0
a=sendonly
a=mid:0
a=rtcp-mux
a=ice-ufrag:iBLd
a=ice-pwd:s31NZ3bzksBs3+DvMAvko9
a=ice-options:trickle
a=fingerprint:sha-256 52:CB:F9:3D:51:E0:9A:94:EC:01:86:F5:11:90:17:05:C3:42:ED:A5:41:27:E2:EA:8A:B1:AC:29:BC:2F:19:C4
a=setup:passive
a=candidate:1 1 udp 2015363327 192.168.119.128 58364 typ host
a=candidate:2 1 udp 1015021823 192.168.119.128 0 typ host
a=candidate:3 1 udp 1010827519 192.168.119.128 58373 typ host
a=candidate:7 1 udp 1679819007 183.134.94.123 59217 typ srflx
a=candidate:8 1 udp 847249663 183.134.94.123 0 typ srflx
a=candidate:9 1 udp 843055359 183.134.94.123 58373 typ srflx

chrome webrtc internal debug：
[image]

robertqd, 2022-04-19:
the issue happened in Chrome  100.0.4896.127（正式版本） （64 位） . there is no issue at the lower version(e.g. 99.xxx) than this version

sepfy, 2022-04-20:
Hi @robertqd.
Which example are you running?

robertqd, 2022-04-21:
hi，@sepfy

thanks for your quick reply.
the root cause has been found.
I run video_on_demand.
For chrome 100.0.4896,127(latest version),  the codec 102 is not found in offer sdp from chrome(deleted by chrome? take place of 127 ?i don't konw why? if you know, could you tell me? thanks).
i only modify the codec number(102) in the sdp answer, but miss to modify the payloadtype  in RTP. The RTP still carry 102 payloadtype. it will cause chrome not display video.

thanks
robert

sepfy, 2022-04-21:
I'm sorry for this issue. Because this library is still under development, many functions are incomplete. In fact, the payload type of codec is different in each browser(Chrome, Firefox... even different platform), so it should not be a fixed value. It should refer the offer and update the RTP packet.
You can modify the payload type in [here](https://github.com/sepfy/pear/blob/main/examples/video_on_demand/h264_packetizer.c#L40) first.

Maybe I can fix this issue in the next few weeks.

huynhphong1611, 2023-02-17:
I got the same error. It doesn't show up on google chrome version 110. please help me !!!

hialle, 2023-02-20:
> I got the same error. It doesn't show up on google chrome version 110. please help me !!!

Hi @huynhphong1611, do you run the example on PC or raspberry pi ?

sepfy, 2023-10-22:
RTP payload type is now consistent with SDP

closed 2023-10-22 by sepfy