sepfy/libpeer #17: PIPE LINE doesn't work well on VM ubuntu

robertqd, 2022-05-01:
Hello @sepfy !

i run the example [surveillance](https://github.com/sepfy/pear/tree/main/examples/surveillance) at my vm ubuntu.
The "PIPE_LINE" doesn't work.  i google some other command and still not work well. i don't familiar with gstreamer comand.
could you provide an comand which can work well ？

thanks
Robert

robertqd, 2022-05-01:
hi @sepfy !

The issue has been found.
use the gstreamer command in surverillance example, the profile of video h264 is high. 
but the profile of sdp negotiation is base. so the video is not decoded in chrome.
add "video/x-264,profile=baseline" into gstreamer comand, it can work well.

thanks
Robert

sepfy, 2022-05-07:
The current pipeline is set for Raspberry Pi. Maybe I can consider changing it to a generic pipeline.
If there are no other problems. We will close the issue.

robertqd, 2022-05-10:
ok, thanks