sepfy/libpeer #20: Running Surveillance Example on Pi3

Vishal5757, 2022-07-03:
Hi Sepfy,
I started of with video on demand , got that to work just after the change was made to the h264_packetizer.c(102 --> 127) . After I went through the issue list.

Now trying the Surveillance example . I am using a C270 Logitech camera connected over USB:

This below are supported formats of camera:
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
        [1]: 'MJPG' (Motion-JPEG, compressed)

The gstream pipeleine setting I use is this:
const char PIPE_LINE[] = "v4l2src device=/dev/video0 ! videoconvert ! video/-raw,width=544,height=288,framerate=10/1 ! v4l2h264enc ! h264parse ! mp4mux ! appsink name=peer-connection-sink";

Just to very above setting work fine dumped stream to mp4 file and it plays back just fine like so:
gst-launch-1.0 -e v4l2src device=/dev/video0 ! videoconvert ! video/-raw,width=544,height=288,framerate=10/1 ! v4l2h264enc ! h264parse ! mp4mux ! filesink location=video.mp4

My problem now is when i run the sample I keep getting this error 

./examples/surveillance/surveillance
Get offer from singaling
[INFO /home/pi/myApp/pear/src/dtls_transport.c:291] Remote fingerprint E1:24:F6:04:0E:80:B2:04:AE:3C:6C:18:1B:CD:48:63:A3:97:82:A1:DC:78:C5:E3:07:6B:70:E3:DA:8D:70:03
[INFO /home/pi/myApp/pear/src/dtls_transport.c:292] Local fingerprint 3C:7D:45:8A:DE:52:26:07:49:89:58:B4:8E:B9:E6:44:5E:CD:D4:60:88:4E:32:ED:9B:5C:8D:0C:FD:62:6C:BA
[INFO /home/pi/myApp/pear/src/dtls_transport.c:331] Created inbound SRTP session
[INFO /home/pi/myApp/pear/src/dtls_transport.c:337] Created outbound SRTP session

(surveillance:1510): GLib-GObject-CRITICAL **: 20:26:01.074: g_object_set: assertion 'G_IS_OBJECT (object)' failed

Any hints!!!!

sepfy, 2022-07-28:
Hello

Maybe try this pipeline?
```
const char PIPE_LINE[] = "v4l2src device=/dev/video0 ! videoconvert ! video/-raw,width=544,height=288,framerate=10/1 ! v4l2h264enc ! h264parse ! rtph264pay name=rtp config-interval=-1 ssrc=1 ! appsink name=peer-connection-sink";
```

Because we should streaming H264 raw data as RTP packet.

sepfy, 2023-10-22:
This example is no longer supported

closed 2023-10-22 by sepfy