sepfy/libpeer #30: I do not connect rtsp use pear rtspsrc location=rtsp

huynhphong1611, 2023-02-24:
I do not connect rtsp use pear rtspsrc location=rtsp .
Can you help me connect rtsp with example?

sepfy, 2023-02-26:
hi, if you want to use rtsp video stream as a source. can try to modify the pipeline of surveillance to
```
rtspsrc location=<rtsp url> ! rtph264depay ! h264parse ! rtph264pay name=rtp config-interval=-1 ssrc=1 ! appsink name=peer-connection-sink
```

huynhphong1611, 2023-02-27:
Thank you ,
I tried and succeeded !

huynhphong1611, 2023-02-27:
Hi, 
I have tried and succeeded with video .
But no sound from rtsp. Although my rtsp server has sound.

Can you help me add this sound? Thank you.

huynhphong1611, 2023-02-27:
I think you should have an example of rtsp using pear.
Because IP cameras mainly use rtsp as a video source.

sepfy, 2023-10-22:
There are currently no plans to support this feature

closed 2023-10-22 by sepfy