sepfy/libpeer #39: CMake Error at src/CMakeLists.txt:29 (install): install TARGETS given no ARCHIVE DESTINATION for static library target. "pear".

xgcaspnetbysj, 2023-07-28:
my cmake version is as follows:
cmake --version
cmake version 3.8.2
CMake suite maintained and supported by Kitware (kitware.com/cmake).

when i run cmake .. , i encounter the error:
CMake Error at src/CMakeLists.txt:29 (install):
  install TARGETS given no ARCHIVE DESTINATION for static library target
  "pear".

-- Configuring incomplete, errors occurred!
See also "/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeOutput.log".

I am looking for your help. Thank you very much !

Here is CMakeOutput.log :
The system is: Linux - 5.4.0-150-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/3.8.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/3.8.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_d0408/fast"
/usr/bin/make -f CMakeFiles/cmTC_d0408.dir/build.make CMakeFiles/cmTC_d0408.dir/build
make[1]: Entering directory '/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d0408.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_d0408.dir/testCCompiler.c.o   -c /home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_d0408
/opt/gcc_720/mips-gcc720-glibc229-r5.1.4/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d0408.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_d0408.dir/testCCompiler.c.o  -o cmTC_d0408 
make[1]: Leaving directory '/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_301c0/fast"
/usr/bin/make -f CMakeFiles/cmTC_301c0.dir/build.make CMakeFiles/cmTC_301c0.dir/build
make[1]: Entering directory '/home/lzy/Downloads/PearWR/pear/cmake/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_301c0.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_301c0.dir/CMakeCCompilerABI.c.o   -c /opt/gcc_720/mips-gcc720-glibc229-r5.1.4/share/cmake-3.8/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_301c0
/opt/gcc_720/mips-gcc720-glibc229-r5.1.4/bin/cmake -E cmake_link_script CMakeFiles/cmTC_301c0.dir/link.txt --verbose=1
/usr/bin/cc     -v -rdynamic CMakeFiles/cmTC_301c0.dir/CMakeCCompilerABI.c.o  -o cmTC_301c0 
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
[... 540 lines cut ...]
    Feature record: CXX_FEATURE:0cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:0cxx_nullptr
    Feature record: CXX_FEATURE:0cxx_override
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:0cxx_right_angle_brackets
    Feature record: CXX_FEATURE:0cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:0cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates

That is all.

sepfy, 2023-07-29:
Hi, could you try to add ARCHIVE DESTINATION to src/CMakeLists.txt?
```
install(TARGETS pear
  LIBRARY DESTINATION ${CMAKE_INSTALL_PREFIX}/lib/
  ARCHIVE DESTINATION ${CMAKE_INSTALL_PREFIX}/lib/
  PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_PREFIX}/include/
)
```

xgcaspnetbysj, 2023-08-08:
When i modify CMakeLists.txt as follows :
`
cmake_minimum_required(VERSION 3.1)
project(pear)

option(ENABLE_TEST "Enable unit test" OFF)
option(ENABLE_EXAMPLES "Build examples" OFF)
option(ENABLE_SIGNALING "Enable signaling service" OFF)

if(ENABLE_EXAMPLES)
  set(ENABLE_SIGNALING ON)
endif()

find_package(PkgConfig)
pkg_check_modules(GLIB REQUIRED glib-2.0)

set(LIBNICE_INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/third_party/libnice/agent)
set(LIBSRTP_INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/third_party/libsrtp/include)
set(LIBRTP_INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/third_party/media-server/librtp/include)
set(USRSCTP_INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/third_party/usrsctp/usrsctplib)
include_directories(${GLIB_INCLUDE_DIRS} ${LIBNICE_INCLUDE_DIRS} ${LIBSRTP_INCLUDE_DIRS} ${LIBRTP_INCLUDE_DIRS} ${USRSCTP_INCLUDE_DIRS})

set(LIBNICE_LINK_DIRS ${CMAKE_SOURCE_DIR}/third_party/libnice/builddir/nice/)
set(LIBSRTP_LINK_DIRS ${CMAKE_SOURCE_DIR}/third_party/libsrtp/build/)
set(LIBRTP_LINK_DIRS ${CMAKE_SOURCE_DIR}/third_party/media-server/librtp/debug.linux)
set(USRSCTP_LINK_DIRS ${CMAKE_SOURCE_DIR}/third_party/usrsctp/build/usrsctplib)
link_directories(${LIBNICE_LINK_DIRS} ${LIBSRTP_LINK_DIRS} ${LIBRTP_LINK_DIRS} ${USRSCTP_LINK_DIRS})

set(TARGET_LIBS "ssl" "crypto" "nice" "srtp2" "rtp" "usrsctp" "pthread" "glib-2.0" "gobject-2.0" "gio-2.0")

if(ENABLE_SIGNALING)
  set(SIGNALING_INCLUDE_DIRS ${CMAKE_SOURCE_DIR}/third_party/httpserver.h ${CMAKE_SOURCE_DIR}/third_party/cJSON)
  include_directories(${SIGNALING_INCLUDE_DIRS})
  set(SIGNALING_LIB_DIRS ${CMAKE_SOURCE_DIR}/third_party/cJSON)
  link_directories(${SIGNALING_LIB_DIRS})
  LIST(APPEND TARGET_LIBS "cjson")
endif()

add_subdirectory(src)

if(ENABLE_EXAMPLES)
add_subdirectory(examples)
endif()

**install(TARGETS pear
		LIBRARY DESTINATION ${CMAKE_INSTALL_PREFIX}/lib/
		ARCHIVE DESTINATION ${CMAKE_INSTALL_PREFIX}/lib/
		PUBLIC_HEADER DESTINATION ${CMAKE_INSTALL_PREFIX}/include/
	)**
`
, then i perform the command : cmake ..  in the current directory, I encounter the error ：
lzy@ubuntu:~/Downloads/PearWR/pear/cmake$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.56.4
CMake Error at src/CMakeLists.txt:29 (install):
[... 57 lines cut ...]
    Feature record: CXX_FEATURE:0cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:0cxx_nullptr
    Feature record: CXX_FEATURE:0cxx_override
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:0cxx_right_angle_brackets
    Feature record: CXX_FEATURE:0cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:0cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    F
a
:ure record: CXX_FEATURE:0cxx_variadic_templates

xgcaspnetbysj, 2023-08-08:
Great ! It work well. Thank you very much!

closed 2023-08-09 by sepfy