sepfy/libpeer #42: Can libpeer support  WHIP/WHEP standard signaling?

chatop2020, 2023-09-13:
Can libpeer support the real -time push flow of WEBRTC in ZLMediakit based on WHIP/WHEP standard signaling?
zlmediakit opensource on github  https://github.com/ZLMediaKit/ZLMediaKit
thanks.

chatop2020, 2023-09-13:
是否可以支持zlmediakit的webrtc服务器？它可以使用http或WHIP/WHEP信令协议，能否尝试兼容一下？

chatop2020, 2023-09-13:
我的主要场景是使用在ESP32上，谢谢了

sepfy, 2023-09-14:
您好, 我會研究下如何在ESP32上兼容WHIP/WHEP信令協議, 如果您的應用場景是將ESP32 Camera推流到流媒体服務器的話, 可能會面臨到編碼格式的問題
https://components.espressif.com/components/espressif/esp_h264

c4b4d4, 2023-09-15:
Hey there I'm also interesting in WHIP. It makes all the handshaking way easier to integrate to third-party media servers. https://www.ietf.org/archive/id/draft-ietf-wish-whip-01.html

chatop2020, 2023-09-15:
> 您好, 我會研究下如何在ESP32上兼容WHIP/WHEP信令協議, 如果您的應用場景是將ESP32 Camera推流到流媒体服務器的話, 可能會面臨到編碼格式的問題 https://components.espressif.com/components/espressif/esp_h264

其实我主要是使用音频，当然H264能支持是最好的了

chatop2020, 2023-09-26:
请问WHIP/WHEP支持现在有进度吗？

sepfy, 2023-09-30:
Hi. 目前[whip](https://github.com/sepfy/libpeer/tree/whip)分支上的[esp32s3](https://github.com/sepfy/libpeer/tree/whip/examples/esp32s3)範例已經能透過WHIP推流到ZLMediakit或Janus (Simple WHIP Server)
但尚未完成HTTPS的支持, 並且受限ESP32的效能, 我只有成功嘗試 OPUS (8000HZ/18000bps)和 H264 (96x96/7FPS),
可以透過修改[config.h](https://github.com/sepfy/libpeer/blob/whip/src/config.h)來設置media server地址

c4b4d4, 2023-09-30:
I have no problem about being HTTP on my side.

I will give it a try next week! Will let you know how it goes.

c4b4d4, 2023-10-05:
Hi,

I've tested the code and I haven't been able to receive the stream correctly. I doubt that my WHIP endpoint is not working correctly, since I can use it with other services.

I debugged a little bit the code and found where is getting stuck:

It's in an infinite loop in this line: https://github.com/sepfy/libpeer/blob/6ec50f7d4d96711a0d26b1888701102703949ae9/src/peer_connection.c#L347

I've added a LOG in this section

```
case PEER_CONNECTION_NEW:
      LOGI("NEW CONNECTION %d",pc->b_offer_created);
      if (!pc->b_offer_created) {
        peer_connection_state_new(pc);
      }
    break;
```

And well, the code is landing infinitely in `PEER_CONNECTION_NEW` and the value of `pc->b_offer_created` is always `true`

```
INFO    /libpeer-seed/libpeer/src/peer_connection.c  352     NEW CONNECTION 1
INFO    /libpeer-seed/libpeer/src/peer_connection.c  352     NEW CONNECTION 1
INFO    /libpeer-seed/libpeer/src/peer_connection.c  352     NEW CONNECTION 1
INFO    /libpeer-seed/libpeer/src/peer_connection.c  352     NEW CONNECTION 1
INFO    /libpeer-seed/libpeer/src/peer_connection.c  352     NEW CONNECTION 1
```

What could cause the code to be infinitely looping in this case? 🤔

**Note:**

I was also trying to add my own `STUN SERVER`, but it fails to be added, and it gets stuck in here:

```
ERROR   /libpeer-seed/libpeer/src/stun.c     196     Unknown Attribute Type: 0x802b
ERROR   /libpeer-seed/libpeer/src/stun.c     196     Unknown Attribute Type: 0x802c
ERROR   /libpeer-seed/libpeer/src/stun.c     196     Unknown Attribute Type: 0x8022
```

These attributes are for:

```
0x802b -> Response Origin
0x802c -> Other Address
0x8022 -> Software
```

c4b4d4, 2023-10-07:
I just increased the timeout at https://github.com/sepfy/libpeer/blob/whip/src/peer_signaling.c#L18 like @sepfy suggested and it's working now 😄

I haven't tried with my own STUN server, will try that now

chatop2020, 2023-10-07:
> Hi. 目前[whip](https://github.com/sepfy/libpeer/tree/whip)分支上的[esp32s3](https://github.com/sepfy/libpeer/tree/whip/examples/esp32s3)範例已經能透過WHIP推流到ZLMediakit或Janus (Simple WHIP Server) 但尚未完成HTTPS的支持, [...]

感谢您的支持，谢谢，我试一下。

chatop2020, 2023-10-09:
作者你好，我尝试在examples/esp32s3和examples/esp32两个目录下运行"idf.py set-target esp32s3"，报错了，提示缺少srtp组件，具体报错信息如下，请帮忙指导一下怎么解决这个问题，谢谢！

> 
 esp32 git:(whip) ✗ idf.py set-target esp32s3
Setting IDF_PATH environment variable: /Users/qiuzhouwei/Sources/esp-idf
/Users/qiuzhouwei/Sources/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/Users/qiuzhouwei/Sources/libpeer/examples/esp32/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory /Users/qiuzhouwei/Sources/libpeer/examples/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/qiuzhouwei/Sources/esp-idf/tools/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /Users/qiuzhouwei/Sources/libpeer/examples/esp32"...
-- Existing sdkconfig '/Users/qiuzhouwei/Sources/libpeer/examples/esp32/sdkconfig' renamed to '/Users/qiuzhouwei/Sources/libpeer/examples/esp32/sdkconfig.old'.
-- Found Git: /opt/homebrew/bin/git (found version "2.39.0") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
....Updating lock file at /Users/qiuzhouwei/Sources/libpeer/examples/esp32/dependencies.lock
Processing 3 dependencies:
[1/3] espressif/esp32-camera (2.0.5)
[2/3] espressif/mdns (1.2.1)
[3/3] idf (5.1.0)
CMake Error at /Users/qiuzhouwei/Sources/esp-idf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'srtp'.
Call Stack (most recent call first):
  /Users/qiuzhouwei/Sources/esp-idf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /Users/qiuzhouwei/Sources/esp-idf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  /Users/qiuzhouwei/Sources/esp-idf/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/qiuzhouwei/Sources/libpeer/examples/esp32/build/CMakeFiles/CMakeOutput.log".
HINT: The component srtp could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.

sepfy, 2023-10-22:
WHIP supported 192d7f0d153ac9b53a6dfc538e10d3e689fecc33

closed 2023-10-22 by sepfy