sepfy/libpeer #45: whip的esp32s3编译报错

chatop2020, 2023-10-09:
作者你好，我测试whip模式，无法通过编译，帮忙看看是否有什么需要注意的地方？

以下是报错信息

[993/1005] Building C object esp-idf/peer/CMakeFiles/.../Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c.obj
FAILED: esp-idf/peer/CMakeFiles/__idf_peer.dir/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c.obj 
/Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DHAVE_CONFIG_H -DIDF_VER=\"v5.1-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/build/config -I/Users/qiuzhouwei/Sources/libpeer/src -I/Users/qiuzhouwei/Sources/libpeer/src/transports -I/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/include -I/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/interface -I/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/dependency/3rdparty/llhttp/include -I/Users/qiuzhouwei/Sources/libpeer/third_party/coreMQTT/source/include -I/Users/qiuzhouwei/Sources/libpeer/third_party/coreMQTT/source/interface -I/Users/qiuzhouwei/Sources/esp-idf/components/newlib/platform_include -I/Users/qiuzhouwei/Sources/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/qiuzhouwei/Sources/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/qiuzhouwei/Sources/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/qiuzhouwei/Sources/esp-idf/components/freertos/esp_additions/include -I/Users/qiuzhouwei/Sources/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_hw_support/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_hw_support/include/soc -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/Users/qiuzhouwei/Sources/esp-idf/components/heap/include -I/Users/qiuzhouwei/Sources/esp-idf/components/log/include -I/Users/qiuzhouwei/Sources/esp-idf/components/soc/include -I/Users/qiuzhouwei/Sources/esp-idf/components/soc/esp32s3 -I/Users/qiuzhouwei/Sources/esp-idf/components/soc/esp32s3/include -I/Users/qiuzhouwei/Sources/esp-idf/components/hal/esp32s3/include -I/Users/qiuzhouwei/Sources/esp-idf/components/hal/include -I/Users/qiuzhouwei/Sources/esp-idf/components/hal/platform_port/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/include/esp32s3 -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3 -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_common/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_system/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_system/port/soc -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_system/port/include/private -I/Users/qiuzhouwei/Sources/esp-idf/components/xtensa/include -I/Users/qiuzhouwei/Sources/esp-idf/components/xtensa/esp32s3/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/include/apps -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/include/apps/sntp -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/lwip/src/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/port/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/port/freertos/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/port/esp32xx/include -I/Users/qiuzhouwei/Sources/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/qiuzhouwei/Sources/esp-idf/components/mbedtls/port/include -I/Users/qiuzhouwei/Sources/esp-idf/components/mbedtls/mbedtls/include -I/Users/qiuzhouwei/Sources/esp-idf/components/mbedtls/mbedtls/library -I/Users/qiuzhouwei/Sources/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/components/srtp/esp-port -I/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/components/srtp/libsrtp/include -I/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/components/srtp/libsrtp/crypto/include -I/Users/qiuzhouwei/Sources/esp-idf/components/json/cJSON -I/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/managed_components/espressif__mdns/include -I/Users/qiuzhouwei/Sources/esp-idf/components/console -I/Users/qiuzhouwei/Sources/esp-idf/components/vfs/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_netif/include -I/Users/qiuzhouwei/Sources/esp-idf/components/esp_event/include -mlongcalls  -fdiagnostics-color=always   -DESP32 -DHTTP_DO_NOT_USE_CUSTOM_CONFIG -DMQTT_DO_NOT_USE_CUSTOM_CONFIG -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3=. -fmacro-prefix-map=/Users/qiuzhouwei/Sources/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/peer/CMakeFiles/__idf_peer.dir/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c.obj -MF esp-idf/peer/CMakeFiles/__idf_peer.dir/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c.obj.d -o esp-idf/peer/CMakeFiles/__idf_peer.dir/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c.obj -c /Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c: In function 'dtls_srtp_selfsign_cert':
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:123:3: error: implicit declaration of function 'mbedtls_x509write_crt_set_serial'; did you mean 'mbedtls_x509write_crt_set_serial_raw'? [-Werror=implicit-function-declaration]
  123 |   mbedtls_x509write_crt_set_serial(&crt, &serial);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   mbedtls_x509write_crt_set_serial_raw
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c: In function 'dtls_srtp_init':
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:147:16: error: unknown type name 'mbedtls_ssl_srtp_profile'
  147 |   static const mbedtls_ssl_srtp_profile default_profiles[] = {
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:148:4: error: 'MBEDTLS_TLS_SRTP_AES128_CM_HMAC_SHA1_80' undeclared (first use in this function)
  148 |    MBEDTLS_TLS_SRTP_AES128_CM_HMAC_SHA1_80,
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:148:4: note: each undeclared identifier is reported only once for each function it appears in
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:149:4: error: 'MBEDTLS_TLS_SRTP_AES128_CM_HMAC_SHA1_32' undeclared (first use in this function)
  149 |    MBEDTLS_TLS_SRTP_AES128_CM_HMAC_SHA1_32,
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:150:4: error: 'MBEDTLS_TLS_SRTP_NULL_HMAC_SHA1_80' undeclared (first use in this function)
  150 |    MBEDTLS_TLS_SRTP_NULL_HMAC_SHA1_80,
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:151:4: error: 'MBEDTLS_TLS_SRTP_NULL_HMAC_SHA1_32' undeclared (first use in this function)
  151 |    MBEDTLS_TLS_SRTP_NULL_HMAC_SHA1_32,
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:152:4: error: 'MBEDTLS_TLS_SRTP_UNSET' undeclared (first use in this function); did you mean 'MBEDTLS_TLS_EXT_SCT'?
  152 |    MBEDTLS_TLS_SRTP_UNSET
      |    ^~~~~~~~~~~~~~~~~~~~~~
      |    MBEDTLS_TLS_EXT_SCT
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:197:5: error: implicit declaration of function 'mbedtls_ssl_conf_dtls_cookies'; did you mean 'mbedtls_ssl_conf_dn_hints'? [-Werror=implicit-function-declaration]
  197 |     mbedtls_ssl_conf_dtls_cookies(&dtls_srtp->conf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &dtls_srtp->cookie_ctx);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     mbedtls_ssl_conf_dn_hints
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:211:3: error: implicit declaration of function 'mbedtls_ssl_conf_dtls_srtp_protection_profiles' [-Werror=implicit-function-declaration]
  211 |   mbedtls_ssl_conf_dtls_srtp_protection_profiles(&dtls_srtp->conf, default_profiles);
[... cut ...]
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c: At top level:
/Users/qiuzhouwei/Sources/libpeer/src/dtls_srtp.c:79:12: warning: 'dtls_srtp_cert_verify' defined but not used [-Wunused-function]
   79 | static int dtls_srtp_cert_verify(void *data, mbedtls_x509_crt *crt, int depth, uint32_t *flags) {
      |            ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[996/1005] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.obj
ninja: build stopped: subcommand failed.

c4b4d4, 2023-10-10:
What steps did you follow?

I only followed these: https://github.com/sepfy/libpeer/tree/whip/examples/esp32#instructions but applied to the esp32s3's example folder (https://github.com/sepfy/libpeer/tree/whip/examples/esp32s3) and it worked

chatop2020, 2023-10-10:
> What steps did you follow?
> 
> I only followed these: https://github.com/sepfy/libpeer/tree/whip/examples/esp32#instructions but applied to the esp32s3's example folder (https://github.com/sepfy/libpeer/tree/whip/examples/esp32s3) and it worked

i'll try it ,thanks alot

chatop2020, 2023-10-10:
I arranged the document method to try, the error is still the same, I mainly want to test the whip function of esp32s3, but the compilation error, can not continue In the compilation process, coreMQTT and coreHTTP are missing, I found the source code on github to supplement, but the compilation error still exists, the following is the complete error information

➜  esp32s3 git:(whip) ✗ idf.py build
Setting IDF_PATH environment variable: /Users/qiuzhouwei/Sources/esp-idf
/Users/qiuzhouwei/Sources/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/qiuzhouwei/Sources/esp-idf/tools/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/sdkconfig'
-- Found Git: /opt/homebrew/bin/git (found version "2.39.0") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/qiuzhouwei/Sources/esp-idf/tools/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 5 dependencies:
[1/5] espressif/esp32-camera (2.0.5)
[2/5] espressif/esp_audio_codec (1.0.1)
[3/5] espressif/esp_h264 (0.1.1)
[4/5] espressif/mdns (1.2.1)
[5/5] idf (5.1.0)
-- Project sdkconfig file /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/qiuzhouwei/Sources/esp-idf/tools/python_env/idf5.1_py3.11_env/bin/python (found version "3.11.4") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp32-peer" version: 6cfde5a-dirty
-- Adding linker script /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/qiuzhouwei/Sources/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
{HTTP_INCLUDE_PUBLIC_DIRS}/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/include/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/interface/Users/qiuzhouwei/Sources/libpeer/third_party/coreHTTP/source/dependency/3rdparty/llhttp/include
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp32-camera espressif__esp_audio_codec espressif__esp_h264 espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table peer perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs srtp tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/qiuzhouwei/Sources/esp-idf/components/app_trace /Users/qiuzhouwei/Sources/esp-idf/components/app_update /Users/qiuzhouwei/Sources/esp-idf/components/bootloader /Users/qiuzhouwei/Sources/esp-idf/components/bootloader_support /Users/qiuzhouwei/Sources/esp-idf/components/bt /Users/qiuzhouwei/Sources/esp-idf/components/cmock /Users/qiuzhouwei/Sources/esp-idf/components/console /Users/qiuzhouwei/Sources/esp-idf/components/cxx /Users/ [... cut ...]
  226 |                                       uint32_t *num_octets_to_output);
      |                                       ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[998/1009] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/build/log/idf_py_stderr_output_55881 and /Users/qiuzhouwei/Sources/libpeer/examples/esp32s3/build/log/idf_py_stdout_output_55881
➜  esp32s3 git:(whip) ✗

chatop2020, 2023-10-19:
ok,i will try it

maherahmed, 2023-11-29:
@sepfy I have tried to fix in compatibility issue with esp-idf(>5.0.2) that you mentioned. 

But I get this error
`ERROR   C:/libPeer-master/libpeer/src/dtls_srtp.c       399     failed! mbedtls_ssl_handshake returned -0x7280`

Any ideas?

sepfy, 2024-09-20:
It should be a problem with using Firefox (6e9cd9cc6d2ed271ce678228d8c84d652ba7dd3e). Close it first. If it still happens again, please reopen it and inform the browser you are using. Thanks

closed 2024-09-20 by sepfy