sepfy/libpeer #46: mdns?

richlegrand, 2023-10-18:
Hello, 
I'm using an esp32-s3 with OV2640 camera.  I have the demo working when I connect from within the local network.  (pointing browser to https://sepfy.github.io/webrtc/?deviceId=esp32-xxxxxxxxxxxxxxxxxx) --- awesome :) 

One problem I'm having is when I try to connect from outside the local network, I get the console message:

WARN	/home/rich/ringtail/libpeer/src/ports.c	118	ESP_ERR_NOT_FOUND: Host was not found!

Which says that it's having trouble resolving the mdns address.  Since I'm connecting from outside the network, it seems like it's confused (?)  What could be happening?

thanks!

sepfy, 2023-10-22:
Yes. when you are using the different local area network, mdns is not available. In your case, was the connection successful? the ice candidate paring (connection of peer to peer) is through the STUN protocol. I will monitor STUN packets to check the connection status

richlegrand, 2023-11-01:
It wasn't able to connect.  What is the best way to debug?
By the way, did you write your own ICE and SCTP code?

thanks

sepfy, 2023-11-05:
yes. I just implemented ice and sctp simply. For Linux os, i am using the USRSCTP. I think the best way is using the wireshark to check the STUN packet has been received and whether the packet is blocked