sepfy/libpeer #47: Error during make

dar8900, 2023-10-20:
I cloned the repository and ran the third party installation script.
I created a build folder and ran `cmake ..` inside it, everything ok.
When I run  the `make` command I find several errors, the first of which is:
``` bash
dtls_srtp.c:62:3: error: unknown type name 'mbedtls_sha256_context'; did you mean 'mbedtls_sha512_context'?
   62 | mbedtls_sha256_context sha256_ctx;
```

the others:
```bash
dtls_srtp.c:239:53: error: unknown type name ‘mbedtls_ssl_key_export_type’; did you mean ‘mbedtls_ssl_export_keys_t’?
  239 | static void dtls_srtp_key_derivation(void *context, mbedtls_ssl_key_export_type secret_type,

dtls_srtp.c:341:51: error: ‘dtls_srtp_key_derivation’ undeclared (first use in this function)
  341 |   mbedtls_ssl_set_export_keys_cb(&dtls_srtp->ssl, dtls_srtp_key_derivation, dtls_srtp);

```

sepfy, 2023-10-22:
Hi, could you check you version of mbedtls? the version of mbedtls in submodules is v3.4.0. Or if you have an older version of mbedtls on your system, is possible to link to wrong version lib?

pcgrenier, 2024-12-29:
Was this ever resolved? Seeing something similar on main?

sepfy, 2024-12-30:
Third-party installation scripts are integrated with cmake. This problem should be solved. If this problem occurs, please reopen it.

closed 2024-12-30 by sepfy