sepfy/libpeer #52: missing headers

richlegrand, 2023-11-02:
Hello, 
I've been trying to compile the latest code (I noticed the esp32s3 example, which is cool :)  

I'm seeing a couple of missing header files that is making the build fail:
src/peer_signaling.c:7:10: fatal error: core_mqtt.h: No such file or directory
    7 | #include <core_mqtt.h>
src/ssl_transport.h:10:10: fatal error: transport_interface.h: No such file or directory
   10 | #include "transport_interface.h"

thanks :)

sepfy, 2023-11-05:
Hi, I tested with my pc, but there is no error of missing header. could you check the coreMQTT source code is in the third_party directory? if yes, could you try to remove your build file of esp32 and rebuild the example?

richlegrand, 2023-11-07:
Ah, I didn't use the --recursive argument when cloning.  You might edit examples/esp32/README.md.

thanks!

sepfy, 2024-08-17:
Fixed

closed 2024-08-17 by sepfy