sepfy/libpeer #57: IPv6 support

richlegrand, 2023-11-19:
I'm noticing that some devices (phones) only have ipv6 addresses available.  What is needed to support ipv6?  Here's what I see -- let me know your thoughts:

1) Add ipv6 support to open, bind, sendto, etc. in udp.c, and address.c.
2) Get ipv6 addresses (local and global?) from device. 
3) ipv6 STUN
4) Generate ipv6 ice candidates and an ipv6 reflexive candidate if needed.
5) Generate candidate pairs by only matching local ipv4s to remote ipv4s and local ipv6s to remote ipv6s.

There are probably other things that I haven't thought of :) 

thanks!

nonibytes, 2024-03-10:
Did you make any progress on this?

sepfy, 2024-06-17:
Added ipv6 support:
https://github.com/sepfy/libpeer/pull/72

closed 2024-06-17 by sepfy