sepfy/libpeer #73: cmake build error, cannot find source file - please help. thank you - linux ubuntu

gokulkiot, 2024-06-17:
`$ cmake ..
-- Configuring done
CMake Error at src/CMakeLists.txt:7 (add_library):
  Cannot find source file:

    /home/vineet/Downloads/vineet/libpeer_new/libpeer/third_party/coreHTTP/source/dependency/3rdparty/llhttp/src/api.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at src/CMakeLists.txt:7 (add_library):
  No SOURCES given to target: peer

CMake Generate step failed.  Build files cannot be regenerated correctly.
`

sepfy, 2024-06-17:
Hello, did you clone with `--recursive` ?

gokulkiot, 2024-06-17:
hey sepfy,
YES, we cloned using --recursive

we followed your instructions in readme

`
$ sudo apt -y install git cmake
$ git clone --recursive https://github.com/sepfy/libpeer
$ cd libpeer
$ ./build-third-party.sh
$ mkdir cmake
$ cd cmake
$ cmake ..
$ make
$ wget http://www.live555.com/liveMedia/public/264/test.264 # Download test video file
$ wget https://mauvecloud.net/sounds/alaw08m.wav # Download test audio file
$ ./examples/sample/sample

`

gokulkiot, 2024-06-17:
Hey now I am stuck with following issue

Processing 5 dependencies:
[1/5] espressif/esp32-camera (2.0.9)
[2/5] espressif/esp_audio_codec (1.0.1)
[3/5] espressif/esp_h264 (0.1.1)
[4/5] espressif/mdns (1.3.2)
[5/5] idf (5.0.2)
CMake Error at /home/gokkri/Downloads/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'srtp'.
Call Stack (most recent call first):
  /home/gokkri/Downloads/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  /home/gokkri/Downloads/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  /home/gokkri/Downloads/vineet/sp-idf/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/gokkri/Downloads/vineet/libpeer_new/libpeer/examples/esp32s3/build/CMakeFiles/CMakeOutput.log".
HINT: The component srtp could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the /home/gokkri/Downloads/vineet/libpeer_new/libpeer/examples/esp32s3/build/log/idf_py_stderr_output_92863 and /home/gokkri/Downloads/vineet/libpeer_new/libpeer/examples/esp32s3/build/log/idf_py_stdout_output_92863
gokkri@gokkri-laptop:~/Downloads/vineet/libpeer_new/libpeer/examples/esp32s3$ 

can you pls look into it

gokulkiot, 2024-06-17:
Hey, thank you very much for the previous reply
now we are stuck with following issue -

Processing 5 dependencies:
[1/5] espressif/esp32-camera (2.0.9)
[2/5] espressif/esp_audio_codec (1.0.1)
[3/5] espressif/esp_h264 (0.1.1)
[4/5] espressif/mdns (1.3.2)
[5/5] idf (5.0.2)
CMake Error at /home/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:245 (message):
Failed to resolve component 'srtp'.
Call Stack (most recent call first):
/home/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
/home/vineet/sp-idf/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
/home/vineet/sp-idf/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/vineet/libpeer_new/libpeer/examples/esp32s3/build/CMakeFiles/CMakeOutput.log".
HINT: The component srtp could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com/' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the /home/vineet/libpeer_new/libpeer/examples/esp32s3/build/log/idf_py_stderr_output_92863 and /home//vineet/libpeer_new/libpeer/examples/esp32s3/build/log/idf_py_stdout_output_92863
$ /Downloads/vineet/libpeer_new/libpeer/examples/esp32s3$

could you pls help us

sepfy, 2024-06-18:
Please follow the instruction as following to download srtp
https://github.com/sepfy/libpeer/blob/main/examples/esp32/README.md

gokulkiot, 2024-06-18:
hey Sepfy, thank you very much for the previous reply.

We are stuck with this error currently

/home/Downloads/vineet/libpeer_new/libpeer/src/dtls_srtp.c: In function 'dtls_srtp_selfsign_cert':
/home/Downloads/vineet/libpeer_new/libpeer/src/dtls_srtp.c:125:3: error: implicit declaration of function 'mbedtls_x509write_crt_set_serial_raw'; did you mean 'mbedtls_x509write_crt_set_serial'? [-Werror=implicit-function-declaration]
  125 |   mbedtls_x509write_crt_set_serial_raw(&crt, (unsigned char*)serial, strlen(serial));
      |   ^~~~~~~~~~~~~~
      |   mbedtls_x509write_crt_set_serial
At top level:
/home/Downloads/vineet/libpeer_new/libpeer/src/dtls_srtp.c:80:12: warning: 'dtls_srtp_cert_verify' defined but not used [-Wunused-function]
   80 | static int dtls_srtp_cert_verify(void *data, mbedtls_x509_crt *crt, int depth, uint32_t *flags) {
      |            ^~~~~~~~~
cc1: some warnings being treated as errors
[895/921] Building C object esp-idf/peer/CMakeFiles/__idf_peer.dir/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c.obj/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c: In function 'peer_signaling_http_post':
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c:236:17: warning: assignment to 'TransportRecv_t' {aka 'long int ()(struct NetworkContext *, void *, unsigned int)'} from incompatible pointer type 'int ()(NetworkContext_t , void *, size_t)' {aka 'int ()(struct NetworkContext *, void *, unsigned int)'} [-Wincompatible-pointer-types]
  236 |   trans_if.recv = ssl_transport_recv;
      |                 ^
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c:237:17: warning: assignment to 'TransportSend_t' {aka 'long int ()(struct NetworkContext *, const void *, unsigned int)'} from incompatible pointer type 'int ()(NetworkContext_t , const void *, size_t)' {aka 'int ()(struct NetworkContext *, const void *, unsigned int)'} [-Wincompatible-pointer-types]
  237 |   trans_if.send = ssl_transport_send;
      |                 ^
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c: In function 'peer_signaling_mqtt_connect':
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c:293:23: warning: assignment to 'TransportRecv_t' {aka 'long int ()(struct NetworkContext *, void *, unsigned int)'} from incompatible pointer type 'int ()(NetworkContext_t , void *, size_t)' {aka 'int ()(struct NetworkContext *, void *, unsigned int)'} [-Wincompatible-pointer-types]
  293 |   g_ps.transport.recv = ssl_transport_recv;
      |                       ^
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c:294:23: warning: assignment to 'TransportSend_t' {aka 'long int ()(struct NetworkContext *, const void *, unsigned int)'} from incompatible pointer type 'int ()(NetworkContext_t , const void *, size_t)' {aka 'int ()(struct NetworkContext *, const void *, unsigned int)'} [-Wincompatible-pointer-types]
  294 |   g_ps.transport.send = ssl_transport_send;
      |                       ^
At top level:
/home/Downloads/vineet/libpeer_new/libpeer/src/peer_signaling.c:230:12: warning: 'peer_signaling_http_post' defined but not used [-Wunused-function]
  230 | static int peer_signaling_http_post(const char *hostname, const char *path, int port, const char *body) {
      |            ^~~~~~~~~~
[902/921] Building C object esp-idf/peer/CMakeFiles/__idf_peer.dir/home/Downloads/vineet/libpeer_new/libpeer/src/sctp.c.obj/home/Downloads/vineet/libpeer_new/libpeer/src/sctp.c:346:12: warning: 'sctp_handle_incoming_data' defined but not used [-Wunused-function]
  346 | static int sctp_handle_incoming_data(Sctp *sctp, char *data, size_t len, uint32_t ppid, uint16_t stream, int flags) {
      |            ^~~~~~~~~
[903/921] Building C object esp-idf/espressif_mdns/CMakeFiles/idf_espressif_mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import mbedtls_x509write_crt_set_serial_raw library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /home/Downloads/vineet/libpeer_new/libpeer/examples/esp32/build/log/idf_py_stderr_output_468285 and /home/Downloads/vineet/libpeer_new/libpeer/examples/esp32/build/log/idf_py_stdout_output_468285

sepfy, 2024-06-30:
Hello, Please update esp-idf to v5.2

closed 2024-08-25 by sepfy