sepfy/libpeer #75: Easy install

karaxuna, 2024-06-29:
Can this work by simply running `idf.py add-dependency sepfy/libpeer`?

sepfy, 2024-08-25:
Hello, I've uploaded the lib to component registry, you can test with `idf.py add-dependency sepfy/libpeer`

karaxuna, 2024-08-28:
@sepfy that's amazing! Only problem is that it does not work with `-DHAVE_USRSCTP` flag:

```
managed_components/sepfy__libpeer/src/sctp.c:7:10: fatal error: usrsctp.h: No such file or directory
    7 | #include <usrsctp.h>
```

sepfy, 2024-08-29:
Hello, Since the ESP32 currently doesn't support using USRSCTP, it defaults to my simplified implementation of SCTP. Thanks to @richlegrand  for helping to port USRSCTP to the ESP32. You can refer to this branch for the progress.
https://github.com/sepfy/libpeer/tree/esp32-usrsctp

karaxuna, 2024-08-29:
That's great! Looking forward to it. I would like to contribute, but I'm
very new to this (I'm a web developer).

karaxuna, 2024-09-03:
Hi, any progress on this? Can I help somehow? It's very critical for me.

sepfy, 2024-09-04:
Hi~ what's feature you need in your project? If you are looking for data channel, libpeer can support it without usrsctp

karaxuna, 2024-09-04:
I have esp32-cam and want to send images via data channel. How to do it
without usrsctp? And then what's the purpose of usrsctp?

sepfy, 2024-09-04:
you can refer this example to send image via data channel:
https://github.com/sepfy/libpeer/tree/main/examples/esp32
In fact, the default implementation uses a simple SCTP implemented by libpeer. While this implementation might lack some features, such as packet loss detection, it can still be used to transmit images. We want to try using usrsctp on the ESP32 to see if it can replace the internal SCTP.

karaxuna, 2024-09-04:
Awesome, it worked! Thank you! I've pushed some prs.

closed 2024-09-17 by karaxuna