sepfy/libpeer #79: mbedtls_ssl_handshake returned -0x7280

magic-tab-01, 2024-07-14:
after run "./examples/sample/sample ", i got error log as pasted, could you kindly help check?

magic@ubuntu:~/work/opensource/libpeer/cmake$ ./examples/sample/sample 
open https://sepfy.github.io/webrtc?deviceId=test_64544
INFO	/home/magic/work/opensource/libpeer/src/peer_connection.c	185	Datachannel allocates heap size: 153600
INFO	/home/magic/work/opensource/libpeer/src/peer_connection.c	190	Audio allocates heap size: 332800
INFO	/home/magic/work/opensource/libpeer/src/peer_connection.c	201	Video allocates heap size: 332800
INFO	/home/magic/work/opensource/libpeer/src/ports.c	159	AF_INET
INFO	/home/magic/work/opensource/libpeer/src/tcp.c	48	port: 8883
INFO	/home/magic/work/opensource/libpeer/src/tcp.c	49	addr: 91.121.93.94
INFO	/home/magic/work/opensource/libpeer/src/tcp.c	55	Connected to server
INFO	/home/magic/work/opensource/libpeer/src/ssl_transport.c	82	start to handshake
INFO	/home/magic/work/opensource/libpeer/src/ssl_transport.c	90	handshake success
INFO	/home/magic/work/opensource/libpeer/src/peer_signaling.c	321	MQTT_Connect succeeded.
INFO	/home/magic/work/opensource/libpeer/src/peer_signaling.c	271	MQTT_PACKET_TYPE_PUBLISH
state is changed: new
INFO	/home/magic/work/opensource/libpeer/src/peer_connection.c	281	ice_servers: stun:stun.l.google.com:19302

INFO	/home/magic/work/opensource/libpeer/src/udp.c	67	local port: 38686
INFO	/home/magic/work/opensource/libpeer/src/agent.c	28	create IPv4 UDP socket: 4
INFO	/home/magic/work/opensource/libpeer/src/agent.c	258	resolved_addr.port: 19302
INFO	/home/magic/work/opensource/libpeer/src/ports.c	159	AF_INET
INFO	/home/magic/work/opensource/libpeer/src/agent.c	265	resolved_addr.ipv4: 74.125.250.129
INFO	/home/magic/work/opensource/libpeer/src/ice.c	83	candidate->addr.family: 2
INFO	/home/magic/work/opensource/libpeer/src/peer_signaling.c	271	MQTT_PACKET_TYPE_PUBLISH
state is changed: checking
state is changed: connected
ERROR	/home/magic/work/opensource/libpeer/src/dtls_srtp.c	376	failed! mbedtls_ssl_handshake returned -0x7280
ERROR	/home/magic/work/opensource/libpeer/src/dtls_srtp.c	376	failed! mbedtls_ssl_handshake returned -0x7280
ERROR	/home/magic/work/opensource/libpeer/src/dtls_srtp.c	376	failed! mbedtls_ssl_handshake returned -0x7280
ERROR	/home/magic/work/opensource/libpeer/src/dtls_srtp.c	376	failed! mbedtls_ssl_handshake returned -0x7280
INFO	/home/magic/work/opensource/libpeer/src/peer_signaling.c	271	MQTT_PACKET_TYPE_PUBLISH
ERROR	/home/magic/work/opensource/libpeer/src/dtls_srtp.c	376	failed! mbedtls_ssl_handshake returned -0x7280
^C^CERROR	/home/magic/work/opensource/libpeer/src/tcp.c	151	Failed to select: Interrupted system call
ERROR	/home/magic/work/opensource/libpeer/src/peer_signaling.c	350	MQTT_ProcessLoop failed: Status=MQTTRecvFailed.

sepfy, 2024-07-15:
hi, Do you encounter this issue every time you run it? Can you confirm that your browser hasn't been refreshed and that your internet connection is stable?

arjo129, 2024-08-31:
I can reproduce this error.

arjo129, 2024-08-31:
Upon further investigation it seems like a firefox related issue. Don't come accross it on Brave.

sepfy, 2024-09-16:
firefox issue fixed
6e9cd9cc6d2ed271ce678228d8c84d652ba7dd3e

closed 2024-09-16 by sepfy