sepfy/libpeer #86: I have a question about PORT.C

topworldcoder, 2024-08-14:
Hi, dear author,

I got candidate like 'a=candidate:2545679721 1 udp 2113937151 b213d6f4-fb35-45e1-ba06-0a276dc6f94c.local 62189 typ host generation 0 network-cost 999'

and libpeer goes to the code in function 'ice_candidate_from_description' (ice.c) and hits 'case 4', then calls 'ports_resolve_mdns_host("b213d6f4-fb35-45e1-ba06-0a276dc6f94c.local", &candidate->addr)'; ->
ports_resolve_addr("b213d6f4-fb35-45e1-ba06-0a276dc6f94c.local", addr); ->
getaddrinfo("b213d6f4-fb35-45e1-ba06-0a276dc6f94c.local", NULL, &hints, &res)

is it correct to call getaddrinfo with content of "b213d6f4-fb35-45e1-ba06-0a276dc6f94c.local" ?
I got an error return: getaddrinfo error: -3

pic from web:
[image]

thanks.

sepfy, 2024-08-17:
It might be that this mDNS cannot be resolved. Are all your devices on the same local network? If not, it's normal that identification fails.

topworldcoder, 2024-08-20:
Thanks for your help.

> It might be that this mDNS cannot be resolved. Are all your devices on the same local network? If not, it's normal that identification fails.

closed 2024-08-20 by topworldcoder