sepfy/libpeer #93: SRTP and MbedTLS dependencies propogate into downstream projects

mitchellhansen-pedantek, 2024-08-19:
Hi!

I have this library integrated into a project of mine, unfortunately to get it to build nicely from an `ExternalProject_Add`, I have to reach into my projects build artifacts to get the headers for `libsrtp` and `mbedtls`

You can see the header chain for mbedtls here...

```
libpeer/src/peer.h:8:
libpeer/src/peer_connection.h:10:
libpeer/src/sctp.h:5:
ibpeer/src/dtls_srtp.h:7:10: fatal error: 'mbedtls/entropy.h' file not found
```

and srtp here

```
libpeer/src/peer.h:8:
libpeer/src/peer_connection.h:10:
libpeer/src/sctp.h:5:
libpeer/src/dtls_srtp.h:7:10: fatal error: 'srtp2/srtp.h' file not found
```

A "Pointer Implementation" strategy is commonly used to avoid this, and was quite easy to add to this project, I have an example PR here that builds very nicely

https://github.com/mitchellhansen-pedantek/libpeer/pull/1

If you find this solution / interface acceptable, I would love to open it for merge.

Thanks!

sepfy, 2024-08-19:
If I directly remove the `#include "sctp.h"`, would you still be able to use it?

```patch
diff --git a/src/peer_connection.c b/src/peer_connection.c
index 14de686..3d9cbfe 100644
--- a/src/peer_connection.c
+++ b/src/peer_connection.c
@@ -164,7 +164,7 @@ PeerConnectionState peer_connection_get_state(PeerConnection *pc) {
   return pc->state;
 }

-Sctp *peer_connection_get_sctp(PeerConnection *pc) {
+void* peer_connection_get_sctp(PeerConnection *pc) {

   return &pc->sctp;
 }
diff --git a/src/peer_connection.h b/src/peer_connection.h
index 1b620e1..e970ffc 100644
--- a/src/peer_connection.h
+++ b/src/peer_connection.h
@@ -7,7 +7,6 @@

 #include <stdlib.h>
 #include <stdint.h>
-#include "sctp.h"

 #ifdef __cplusplus
 extern "C" {
@@ -78,7 +77,7 @@ const char* peer_connection_state_to_string(PeerConnectionState state);

 PeerConnectionState peer_connection_get_state(PeerConnection *pc);

-Sctp *peer_connection_get_sctp(PeerConnection *pc);
+void* peer_connection_get_sctp(PeerConnection *pc);

 PeerConnection* peer_connection_create(PeerConfiguration *config);
 ```

mitchellhansen-pedantek, 2024-08-20:
That worked perfectly, thanks @sepfy 

I created a PR for your convenience 
https://github.com/sepfy/libpeer/pull/95

sepfy, 2024-08-20:
great! thank you

closed 2024-08-20 by sepfy