sepfy/libpeer #96: Function ice_candidate_create got ICE witch TURN relay addr not correct and so on

topworldcoder, 2024-08-20:
Hi, dear author,
    I received candidates from TURN server in function 'ice_candidate_create', original contents may be like 'a=candidate:215664028 1 udp 33562623 xxx.xxx.xxx.xxx 57669 typ relay raddr ccc.ccc.ccc.ccc rport 58862 generation 0 network-cost 999'.

    But function 'agent_create_turn_addr' only use recv_msg.relayed_addr to assign to turn_addr, and pass through to function 'ice_candidate_create' to save as:'memcpy(&candidate->addr, addr, sizeof(Address));'. 

    You can not find any code in ice.c to assign a value to candidate->raddr, but function 'ice_candidate_to_description' would use it. So, I can see the libpeer always parsed and used candidate as 'a=candidate:1 1 UDP 1688684031 xxx.xxx.xxx.xxx 57669 typ relay raddr 0.0.0.0 rport 0' with no mapping address.

    I tried to modify the code at function 'agent_create_turn_addr' like:

static int agent_create_turn_addr(Agent *agent, Address *serv_addr, const char *username, const char *credential) {
codes...
  Address turn_relayaddr;
  Address turn_mappedaddr;
codes...
  memset(&recv_msg, 0, sizeof(recv_msg));
  retry = 0;
  while (++retry < 60) {
      ret = agent_socket_recv(agent, NULL, recv_msg.buf, sizeof(recv_msg.buf));
      if (ret > 0) {
          break;
      }
  }
codes...
  stun_parse_msg_buf(&recv_msg);
  memcpy(&turn_relayaddr, &recv_msg.relayed_addr, sizeof(Address));
  memcpy(&turn_mappedaddr, &recv_msg.mapped_addr, sizeof(Address));
  IceCandidate *ice_candidate = agent->local_candidates + agent->local_candidates_count++;
  ice_candidate_create(ice_candidate, agent->local_candidates_count, ICE_CANDIDATE_TYPE_RELAY, &turn_relayaddr, &turn_mappedaddr);
  return ret;
}

and modify function 'ice_candidate_create' to: assign turn_relayaddr to candidate->raddr, assign turn_mappedaddr to candidate->addr 
Above solved to get the correct info maps. but more issues blocked to continue using TURN and I will show up later.

topworldcoder, 2024-08-20:
static int agent_create_turn_addr(Agent *agent, Address *serv_addr, const char *username, const char *credential) {

  LOGE("Send TURN Binding Request.");
  int ret = -1;
  uint32_t attr = ntohl(0x11000000);
  int retry = 0;
  Address turn_relayaddr;
  Address turn_mappedaddr;
  StunMessage send_msg;
  StunMessage recv_msg;
  memset(&recv_msg, 0, sizeof(recv_msg));
  memset(&send_msg, 0, sizeof(send_msg));
  stun_msg_create(&send_msg, STUN_METHOD_ALLOCATE);
  stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_REQUESTED_TRANSPORT, sizeof(attr), (char*)&attr); // UDP
  stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_USERNAME, strlen(username), (char*)username);

  ret = agent_socket_send(agent, serv_addr, send_msg.buf, send_msg.size);
  if (ret == -1) {
    LOGE("Failed to send TURN Binding Request.");
    return -1;
  }

  // blocking 1 second
  retry = 0;
  while (++retry < 1000) {
    ret = agent_socket_recv(agent, NULL, recv_msg.buf, sizeof(recv_msg.buf));
    if (ret > 0) {
      break;
    }
  }

  if (ret <= 0) {
    LOGD("Failed to receive TURN Binding Response.");
    return ret;
  }

  stun_parse_msg_buf(&recv_msg);

  if (recv_msg.stunclass == STUN_CLASS_ERROR && recv_msg.stunmethod == STUN_METHOD_ALLOCATE) {

    memset(&send_msg, 0, sizeof(send_msg));
    stun_msg_create(&send_msg, STUN_METHOD_ALLOCATE);
    stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_REQUESTED_TRANSPORT, sizeof(attr), (char*)&attr); // UDP
    stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_USERNAME, strlen(username), (char*)username);
    stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_NONCE, strlen(recv_msg.nonce), recv_msg.nonce);
    stun_msg_write_attr(&send_msg, STUN_ATTR_TYPE_REALM, strlen(recv_msg.realm), recv_msg.realm);
    stun_msg_finish(&send_msg, STUN_CREDENTIAL_LONG_TERM, credential, strlen(credential));
  } else {
    LOGE("Invalid TURN Binding Response.");
    return -1;
  }

  ret = agent_socket_send(agent, serv_addr, send_msg.buf, send_msg.size);
  if (ret < 0) {
    LOGE("Failed to send TURN Binding Request.");
    return -1;
  }

  memset(&recv_msg, 0, sizeof(recv_msg));
  retry = 0;
  while (++retry < 60) {
      ret = agent_socket_recv(agent, NULL, recv_msg.buf, sizeof(recv_msg.buf));
      if (ret > 0) {
          break;
      }
  }
  if (ret <= 0) {
    LOGD("Failed to receive TURN Binding Response.");
    return ret;
  }

  stun_parse_msg_buf(&recv_msg);
  memcpy(&turn_relayaddr, &recv_msg.relayed_addr, sizeof(Address));
  memcpy(&turn_mappedaddr, &recv_msg.mapped_addr, sizeof(Address));
  IceCandidate *ice_candidate = agent->local_candidates + agent->local_candidates_count++;
  ice_candidate_create(ice_candidate, agent->local_candidates_count, ICE_CANDIDATE_TYPE_RELAY, &turn_relayaddr, &turn_mappedaddr);

  uint8_t description[CONFIG_MTU];
  ice_candidate_to_description(ice_candidate, (char*)description, CONFIG_MTU);
  LOGD("agent_create_turn_addr get candidate:%s.", description);
  return ret;
}

sepfy, 2024-08-25:
I don't think we need to add `radar` and `report` to the description, but we may have to wait a little longer for a response from TURN. I updated some code, would you mind testing it out? Thanks!
https://github.com/sepfy/libpeer/tree/turn

closed 2024-08-28 by topworldcoder