sepfy/libpeer #97: Corturn's TURN can not get through

topworldcoder, 2024-08-21:
Hi, dear author,
    I got another issue after issue #96.
    For example, I assigned two sets of ice_server in the structure PeerConfiguration, one set is TURN and the other set is STUN, but when calling the function peer_connection_state_new, A loop is used to poll ice_servers, and agent_gather_candidate is called.
    The function agent_gather_candidate uses 'memset(agent, 0, sizeof(Agent));' every time. So even if I assigned two sets of ice_server in the structure PeerConfiguration, can it be considered that the code will always use the last ice_server as the connection and communication, and generate a single Agent *agent (hereinafter referred to as latest_agent).
    In function agent_connectivity_check, the latest_agent will be used to send the binding request (send the IP and port information to the remote candidate).

    I tried many combinations, like:	  
    {.urls = "stun:xxx.xxx.xxx.xxx:5999" },
    {.urls = "turn:xxx.xxx.xxx.xxx:5999", .username = "aaaaaa", .credential = "bbbbbbbbb"},
    or the opposite

    But the embedded device could not connect to the web page opened by the 5G network at all(But another webrtc demo can connect OK). The embedded device could successfully connect to the web page opened by another Internet network.
    (agent_socket_send sent agent_create_binding_request, but agent_recv did not get STUN_METHOD_BINDING by agent_process_stun_response). 

    Is it because I am using Corturn's STUN/TURN that libPeer not supported?

topworldcoder, 2024-08-22:
Web Connect OK with an Internet network, libpeer log:
[image]

Web Connect Failed with a 5G network, libpeer log(with a lot of STUN_METHOD_BINDING request, and finally failed):
[image]

sepfy, 2024-08-25:
hi, is it possible IPv6 issue? maybe you can share some candidates information to me?  i tested with my 5G, TURN server is working for me

topworldcoder, 2024-08-26:
> hi, is it possible IPv6 issue? maybe you can share some candidates information to me? i tested with my 5G, TURN server is working for me

I used the newest codes of libpeer, but libpeer still cannot connect to the web-demo opened with 5G network. My great honor to have your help.

Candidates used by libpeer:
a=candidate:1 1 UDP 1689689855 122.152.232.199 46750 typ srflx raddr 0.0.0.0 rport 0
a=candidate:1 1 UDP 11989503 122.152.232.199 46833 typ relay raddr 0.0.0.0 rport 0

Candidates from web-demo:
a=candidate:1106797033 1 udp 2113937151 ec9cfb17-ce74-42cf-b115-d99f97d9fa93.local 55602 typ host generation 0 network-cost 999
a=candidate:3112090627 1 udp 2113939711 d4c11be6-b478-40fa-82ad-160e9fd26ffb.local 55603 typ host generation 0 network-cost 999
a=candidate:3809684957 1 udp 1677729535 112.97.55.41 28862 typ srflx raddr 0.0.0.0 rport 0 generation 0 network-cost 999
a=candidate:3906187020 1 udp 33562623 122.152.232.199 59970 typ relay raddr 112.97.55.41 rport 28862 generation 0 network-cost 999

And I'd like to send you my TURN server infos if you need too, you can send me an email to timeturnal@outlook.com, and I will reply it.

Pics:
[image]

topworldcoder, 2024-08-27:
My PC connected by 5G can telnet 122.152.232.199 9015, also Linux with libpeer can telnet 122.152.232.199 9015 too.

sepfy, 2024-08-27:
Hi, are the PC and linux with libpeer are the same network? I may need more information about your environment. I've sent my questions to you via email.

topworldcoder, 2024-08-27:
Hi, libpeer runs on the STUN/TURN server system of Linux(which ip is 122.152.232.199, same as TURN's ip), the web-demo runs on 5G network PC(collected by web-demo candidates, which is 223.104.68.137)

STUN/TURN infos I used on both libpeer and web-demo:
PeerConfiguration config = {
 .ice_servers = {
    {.urls = "stun:122.152.232.199:9015" },
    {.urls = "turn:122.152.232.199:9015", .username = "ivyturnsrv", .credential = "ivysrvturn001"},
 },
 .datachannel = DATA_CHANNEL_STRING,
 .video_codec = CODEC_H264,
 .audio_codec = CODEC_PCMA
};

topworldcoder, 2024-08-27:
libpeer is runing now, I start it up as:
./sample ABCDEZZZZF
you can try using ABCDEZZZZF and add TURN infos as above to web-demo

sepfy, 2024-08-27:
Sorry, I tried, but it seems there is no problem with my 5G network. my web demo can connect your server. if you removed the TURN server in the libpeer, is it working?

topworldcoder, 2024-08-28:
> Sorry, I tried, but it seems there is no problem with my 5G network. my web demo can connect your server. if you removed the TURN server in the libpeer, is it working?

I figured out.
This is a very strange thing. Before this, I never doubted that there would be anything wrong with libpeer and STUN/TURN services running on the same server. However, with your reply and my subsequent verification, I got my verification results:

1. libpeer and STUN/TURN services are running on the same linux machine, and web demo is on another Win PC:
libpeer and web demo always fail to connect (connection fails when STUN or TURN is configured independently, and connection fails when STUN/TURN is configured together)
(I have been stuck here before, and I have always suspected that there are problems in various codes or configurations. I don’t know if this scenario also needs to confirm the cause of the problem and subsequent solutions)

2. libpeer is built and run on a server independent of the STUN/TURN service, and web demo is on another Win PC:
libpeer and web demo are successfully connected (running well except firefox)

3. libpeer runs on the VMware virtual machine linux system, web demo runs on the Win PC with the virtual machine that libpeer runned on, and the STUN/TURN service is on another linux system:
libpeer can only connect via TURN, otherwise the connection fails (this makes me wonder, libpeer and web demo should belong to the same LAN, why can't they connect directly but use TURN)

4. libpeer runs on VMware virtual machine linux system, web demo runs on another linux machine, and STUN/TURN service is located on another linux system:
When web demo runs in Firefox, it always reports an error: dtls_srtp.c 376 failed! mbedtls_ssl_handshake returned -0x7280
When web demo runs in Google Chrome, the connection is successful!

5. libpeer is built on the same VMware virtual machine linux system as web demo and runs, and STUN/TURN service is located on another linux system:
same as point 4.

topworldcoder, 2024-08-30:
> ok. so only case 1 and 2 have issue, right? I noticed that the candidate information you provided earlier did not include the host address. I'm not sure if this could cause an issue. Could you please check the network interface configuration in the config.h file? https://github.com/sepfy/libpeer/blob/main/src/config.h#L23

Yes, nothing change at my src/config.h. and What I mean 5G is 'Mobile hotspot of 5G SIM card', not '5G from wifi router'.

rjjrbatarao, 2024-09-06:
I have random success rate with Coturn and I've discovered Eturnal works perfectly in my case you can try
curl -fsS https://eturnal.net/install | sudo sh
- /opt/eturnal
- /etc/eturnal.yml
- /etc/systemd/system/eturnal.service

sudo nano /etc/eturnal.yml
relay_ipv4_addr: Public IPv4 address of the Ec2
relay_ipv6_addr: Public IPv6 address of the Ec2

also add this
credentials:
  alice: l0vesBob
  bob: l0vesAlice
  eve: stalksTh3m