sepfy/libpeer #119: esp32 sample won't connect to the browser

anpin, 2024-09-03:
Hello, I'm evaluating this library for a project. When I run the esp32 sample it wouldn't connect to the frontend. It actually connected the first time, but any further attempts failed. 

[image]

My esp32 and laptop are on the same network, both are right next to a router.

sepfy, 2024-09-04:
hi, i guess you can try to refresh the web page? because web page will send request to device when initialize. you need wait device is connected to server, then open the browser

anpin, 2024-09-04:
I tried to restart the browser and also different browsers and devices. Once I saw an STUN log message after the lines there, but that’s it.

I also try to change device / client id to a random MAC address value.

Could this be limitation of public ally accessible services used by the sample?

sepfy, 2024-09-06:
looks the signaling server is not working for you. Are you able to change the MQTT broker for testing?
https://github.com/sepfy/libpeer/blob/main/src/peer_signaling.h#L25

sepfy, 2024-09-09:
Hello, i checked your log again and i found your peer signaling task is not running (also camera task). you can try to reduce stack size.

anpin, 2024-09-12:
hi @sepfy, thanks for your suggestion, I've tried the updated example from https://github.com/sepfy/libpeer/commit/b9ad64241793c70cceffc7eea77b87949a50ea59 and signaling now works fine.

closed 2024-09-12 by anpin