sepfy/libpeer #126: ZLMediaKit

lilei-John, 2024-09-10:
Can I use the web playback of this open-source project? Can your webRTC only connect to your web?

Sean-Der, 2024-09-11:
ZLMediaKit should work! I haven't done it myself though

Others have done it https://github.com/sepfy/libpeer/issues/42

lilei-John, 2024-09-14:
#define SERVICE_CONFIG_DEFAULT()                                                            \
    {.mqtt_url  = "",                                                                       \
     .mqtt_port = 0,                                                                        \
     .client_id = "peer",                                                                   \
     .http_url  = "default.zlmediakit.com/index/api/webrtc?app=live&stream=test&type=play", \
     .http_port = 443,                                                                      \
     .username  = "",                                                                       \
     .password  = "",                                                                       \
     .pc        = NULL}

./examples/generic/sample 123456
open https://sepfy.github.io/webrtc?deviceId=123456
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   175     Datachannel allocates heap size: 153600
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   180     Audio allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   191     Video allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    500     HTTP Host: default.zlmediakit.com, Port: 443, Path: /index/api/webrtc?app=live&stream=test&type=play
WARN    /home/work/local/webrtc/libpeer/src/peer_signaling.c    447     Invalid MQTT port number: 0
state is changed: new
DEBUG   /home/work/local/webrtc/libpeer/src/dtls_srtp.c 178     local fingerprint: DB:FD:36:33:DC:DC:BA:2E:0E:95:F0:F8:C0:ED:D1:DD:BE:06:39:FC:C4:EB:D3:7A:0D:F9:F1:89:5C:06:E3:72
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   273     ice_servers: stun:stun.l.google.com:19302
INFO    /home/work/local/webrtc/libpeer/src/agent.c     28      create IPv4 UDP socket: 3
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/work/local/webrtc/libpeer/src/agent.c     262     stun/turn server 74.125.250.129:19302
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     265     create stun addr
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      118     XOR Mapped Address Family: 1
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      119     XOR Mapped Address Port: 25700 (Port XOR: 6464)
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      120     XOR Mapped Address IP: 218.205.230.60 (IP XOR: 3ce6cdda)
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     295     local description:
a=ice-ufrag:lDRO
a=ice-pwd:lDROG0vBnc9HBWEqxbBQMgv5
a=candidate:1 1 UDP 2122698239 192.168.1.104 34253 typ host
a=candidate:2 1 UDP 1684301055 218.205.230.60 25700 typ srflx raddr 0.0.0.0 rport 0

INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved default.zlmediakit.com -> 192.168.1.104
INFO    /home/work/local/webrtc/libpeer/src/socket.c    239     Connecting to server: 192.168.1.104:443
INFO    /home/work/local/webrtc/libpeer/src/socket.c    245     Server is connected
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     84      start to handshake
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     92      handshake success
ERROR   /home/work/local/webrtc/libpeer/src/peer_signaling.c    275     Response body is NULL

[image]

I am using Ubuntu virtual machine for testing. Currently, Ubuntu and PC are on the same LAN. How should I use which for testing?

sepfy, 2024-09-15:
libpeer support WHIP API of ZLMediakit, but not support private signaling of ZLMediakit
https://zlmediakit.com/index/api/whip?app=live&stream=test
Herei s the example to connect with ZLMediakit
```
diff --git a/examples/generic/main.c b/examples/generic/main.c
index a49d57e..bc6f825 100644
--- a/examples/generic/main.c
+++ b/examples/generic/main.c
@@ -68,11 +68,6 @@ int main(int argc, char* argv[]) {
   pthread_t peer_singaling_thread;
   pthread_t peer_connection_thread;

-  if (argc < 2) {
-    printf("Usage: %s <device_id>\n", argv[0]);
-    return -1;
-  }
-
   signal(SIGINT, signal_handler);

   PeerConfiguration config = {
@@ -92,13 +87,12 @@ int main(int argc, char* argv[]) {
   peer_connection_oniceconnectionstatechange(g_pc, onconnectionstatechange);
   peer_connection_ondatachannel(g_pc, onmessage, onopen, onclose);

-  service_config.client_id = argv[1];
   service_config.pc = g_pc;
+  service_config.http_url = "127.0.0.1/index/api/whip?app=live&stream=test";
   peer_signaling_set_config(&service_config);
-  peer_signaling_join_channel();
+  peer_signaling_whip_connect();

   pthread_create(&peer_connection_thread, NULL, peer_connection_task, NULL);
-  pthread_create(&peer_singaling_thread, NULL, peer_singaling_task, NULL);

   reader_init();

diff --git a/src/config.h b/src/config.h
index 7b7cdf9..95ebc9b 100644
--- a/src/config.h
+++ b/src/config.h
@@ -17,7 +17,7 @@
 #endif

 #define AUDIO_LATENCY 20  // ms
-#define KEEPALIVE_CONNCHECK 10000
+#define KEEPALIVE_CONNCHECK 0
 #define CONFIG_IPV6 0
```

lilei-John, 2024-09-18:
[image]
./examples/generic/sample 
open https://sepfy.github.io/webrtc?deviceId=(null)
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   175     Datachannel allocates heap size: 153600
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   180     Audio allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   191     Video allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    500     HTTP Host: 192.168.1.104, Port: 443, Path: /index/api/whip?app=live&stream=test
DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    510     MQTT Host: broker.emqx.io, Port: 0
state is changed: new
DEBUG   /home/work/local/webrtc/libpeer/src/dtls_srtp.c 178     local fingerprint: A6:7E:07:95:37:CB:A2:05:81:0A:1F:8C:C9:4B:A0:97:08:CD:59:E9:54:8B:B4:45:DB:E0:F6:A3:EE:83:95:3C
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   273     ice_servers: stun:stun.l.google.com:19302
INFO    /home/work/local/webrtc/libpeer/src/agent.c     28      create IPv4 UDP socket: 3
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/work/local/webrtc/libpeer/src/agent.c     262     stun/turn server 74.125.250.129:19302
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     265     create stun addr
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      118     XOR Mapped Address Family: 1
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      119     XOR Mapped Address Port: 37588 (Port XOR: 92d4)
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      120     XOR Mapped Address IP: 119.4.64.156 (IP XOR: 9c400477)
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     295     local description:
a=ice-ufrag:A30y
a=ice-pwd:A30ynhAFhpBmeG5pSoRMCibS
a=candidate:1 1 UDP 2122922751 192.168.1.104 35130 typ host
a=candidate:2 1 UDP 1687344383 119.4.64.156 37588 typ srflx raddr 0.0.0.0 rport 0

INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved 192.168.1.104 -> 192.168.1.104
INFO    /home/work/local/webrtc/libpeer/src/socket.c    239     Connecting to server: 192.168.1.104:443
INFO    /home/work/local/webrtc/libpeer/src/socket.c    245     Server is connected
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     84      start to handshake
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     92      handshake success
ERROR   /home/work/local/webrtc/libpeer/src/peer_signaling.c    275     Response body is NULL

I have updated the patch file, but it still cannot be played at the moment.

lilei-John, 2024-09-18:
[image]
[image]

sepfy, 2024-10-10:
It looks like ZLMediakit is not responding. You may not have properly enabled the WebRTC functionality. Please refer to the Wiki provided by ZLMediakit to enable WebRTC and try again.
https://github.com/ZLMediaKit/ZLMediaKit/wiki/zlm%E5%90%AF%E7%94%A8webrtc%E7%BC%96%E8%AF%91%E6%8C%87%E5%8D%97

lilei-John, 2024-10-12:
Now it can be played.

closed 2024-10-12 by lilei-John