sepfy/libpeer #128: https://sepfy.github.io/webrtc/?deviceId=12345678 无法播放

lilei-John, 2024-09-13:
浏览器日志
[image]

https://sepfy.github.io/webrtc/?deviceId=12345678
./examples/generic/sample 12345678
open https://sepfy.github.io/webrtc?deviceId=12345678
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   175     Datachannel allocates heap size: 153600
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   180     Audio allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   191     Video allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved broker.emqx.io -> 35.172.255.228
INFO    /home/work/local/webrtc/libpeer/src/socket.c    239     Connecting to server: 35.172.255.228:8883
INFO    /home/work/local/webrtc/libpeer/src/socket.c    245     Server is connected
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     84      start to handshake
ERROR   /home/work/local/webrtc/libpeer/src/ssl_transport.c     88      ssl handshake error: -0x7280
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     92      handshake success
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    356     MQTT_Connect succeeded.
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    298     MQTT_PACKET_TYPE_PUBLISH
state is changed: new
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   273     ice_servers: stun:stun.l.google.com:19302
INFO    /home/work/local/webrtc/libpeer/src/agent.c     28      create IPv4 UDP socket: 4
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/work/local/webrtc/libpeer/src/agent.c     262     stun/turn server 74.125.250.129:19302
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    298     MQTT_PACKET_TYPE_PUBLISH
INFO    /home/work/local/webrtc/libpeer/src/mdns.c      154     Resolved ca6e711c-331a-42e4-9ed9-3c0fbfa42891.local -> 192.168.1.104
INFO    /home/work/local/webrtc/libpeer/src/mdns.c      164     Failed to resolve hostname
state is changed: checking
state is changed: failed

sepfy, 2024-09-13:
could you enable the debug log?
https://github.com/sepfy/libpeer/blob/main/src/config.h#L25

lilei-John, 2024-09-14:
open https://sepfy.github.io/webrtc?deviceId=12345
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   175     Datachannel allocates heap size: 153600
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   180     Audio allocates heap size: 332800
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   191     Video allocates heap size: 332800
DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    510     MQTT Host: broker.emqx.io, Port: 8883
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved broker.emqx.io -> 35.172.255.228
INFO    /home/work/local/webrtc/libpeer/src/socket.c    239     Connecting to server: 35.172.255.228:8883
INFO    /home/work/local/webrtc/libpeer/src/socket.c    245     Server is connected
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     84      start to handshake
ERROR   /home/work/local/webrtc/libpeer/src/ssl_transport.c     88      ssl handshake error: -0x7280
INFO    /home/work/local/webrtc/libpeer/src/ssl_transport.c     92      handshake success
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    356     MQTT_Connect succeeded.
DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    388     MQTT Subscribe/Unsubscribe succeeded.
DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    303     MQTT_PACKET_TYPE_SUBACK
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    298     MQTT_PACKET_TYPE_PUBLISH
state is changed: new
DEBUG   /home/work/local/webrtc/libpeer/src/dtls_srtp.c 178     local fingerprint: AB:F1:E2:A6:43:6E:95:83:0A:17:1A:D1:DF:A2:DA:9A:01:41:8B:AA:C8:71:DC:2F:7A:D1:83:E3:BE:15:CF:88
INFO    /home/work/local/webrtc/libpeer/src/peer_connection.c   273     ice_servers: stun:stun.l.google.com:19302
INFO    /home/work/local/webrtc/libpeer/src/agent.c     28      create IPv4 UDP socket: 4
INFO    /home/work/local/webrtc/libpeer/src/ports.c     161     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/work/local/webrtc/libpeer/src/agent.c     262     stun/turn server 74.125.250.129:19302
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     265     create stun addr
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      95      XOR Mapped Address Family: 0x02
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      96      XOR Mapped Address Port: 50436
DEBUG   /home/work/local/webrtc/libpeer/src/stun.c      97      XOR Mapped Address Address: 203.166.220.189
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     295     local description:
a=ice-ufrag:A3Cw
a=ice-pwd:A3CwgmGnuYoWFmybw904Pd4d
a=candidate:1 1 UDP 1690633471 203.166.220.189 50436 typ srflx raddr 0.0.0.0 rport 0

DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    90      MQTT_Publish succeeded.
INFO    /home/work/local/webrtc/libpeer/src/peer_signaling.c    298     MQTT_PACKET_TYPE_PUBLISH
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     394     Set remote description:
v=0
o=- 7761821140297547852 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE video audio datachannel
a=msid-semantic: WMS
m=video 52303 UDP/TLS/RTP/SAVPF 96 102
c=IN IP4 203.166.220.189
a=rtcp:9 IN IP4 0.0.0.0
a=candidate:2080689296 1 udp 2113937151 77809fd3-6d2b-4bf8-8413-7d95d12039fa.local 52303 typ host generation 0 network-cost 999
a=candidate:4144412988 1 udp 2113939711 e63bd998-4526-4be2-b0c4-756c63d9b2ec.local 52304 typ host generation 0 network-cost 999
a=candidate:2935421366 1 udp 1677732095 2001:470:f8a4:56:2835:61e8:f029:bbd2 52304 typ srflx raddr :: rport 0 generation 0 network-cost 999
a=candidate:2318420204 1 udp 1677729535 203.166.220.189 52303 typ srflx raddr 0.0.0.0 rport 0 generation 0 network-cost 999
a=ice-ufrag:3Max
a=ice-pwd:KhTGIMa7wgpbxd9sYp3ddc46
a=ice-options:trickle
a=fingerprint:sha-256 20:79:EF:F1:85:5C:8B:E8:E8:44:CE:F9:60:D5:A0:E0:AC:29:E5:F3:79:53:A3:22:2E:C0:63:0B:69:AC:FA:C6
a=setup:active
a=mid:video
a=recvonly
a=rtcp-mux
a=rtpmap:96 H264/90000
a=fmtp:96 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42e01f
a=rtpmap:102 H264/90000
a=rtcp-fb:102 nack
a=rtcp-fb:102 nack pli
a=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f
m=audio 9 UDP/TLS/RTP/SAVP 8
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=ice-ufrag:3Max
a=ice-pwd:KhTGIMa7wgpbxd9sYp3ddc46
a=ice-options:trickle
a=fingerprint:sha-256 20:79:EF:F1:85:5C:8B:E8:E8:44:CE:F9:60:D5:A0:E0:AC:29:E5:F3:79:53:A3:22:2E:C0:63:0B:69:AC:FA:C6
a=setup:active
a=mid:audio
a=recvonly
[... 13 lines cut ...]
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      158     timeout
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      158     timeout
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      158     timeout
INFO    /home/work/local/webrtc/libpeer/src/mdns.c      164     Failed to resolve hostname
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      158     timeout
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      158     timeout
DEBUG   /home/work/local/webrtc/libpeer/src/mdns.c      53      flag is not a DNS response
[... cut ...]
DEBUG   /home/work/local/webrtc/libpeer/src/peer_signaling.c    90      MQTT_Publish succeeded.
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     447     send binding request to remote ip: 203.166.220.189, port: 52303
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     447     send binding request to remote ip: 203.166.220.189, port: 52303
DEBUG   /home/work/local/webrtc/libpeer/src/agent.c     447     send binding request to remote ip: 203.166.220.189, port: 52303
state is changed: failed

lilei-John, 2024-09-14:
[image]

lilei-John, 2024-09-14:
It's strange that I switched to a different online environment and now have videos.Is it impossible to penetrate the network?But I see that the browser keeps receiving media data.

sepfy, 2024-09-14:
From your log, it seems that the connection between the two ends was not successfully established. Are you running libpeer and the browser on the same PC? It looks like your case is trying to establish a connection using a public address, but it should be using the local host address. I checked your libpeer logs, and it hasn't generated a host address. Could you check your network interface name and modify it here?
https://github.com/sepfy/libpeer/blob/main/src/config.h#L23
But aside from this issue, I'm not sure why the mDNS resolution failed.

lilei-John, 2024-09-14:
Thank you for your reply. When I return to that network environment again, I will conduct testing.

closed 2024-09-18 by lilei-John