sepfy/libpeer #129: WebRTC connection moves to `PeerConnectionState: 5 (Completed)` but no video after upgrading to ESP-IDF v5.3.1

BehicKlncky, 2024-09-13:
Hi,

I successfully managed to run WebRTC on Xiao ESP32 Sense with libpeer on ESP-IDF v5.2.2. However, after upgrading ESP-IDF to v5.3.1, the connection state changes to `PeerConnectionState: 5` (Completed), but the video stream does not appear.

#### Steps to Reproduce:

- Clone the repo.
- Set up WebRTC signaling and configure for Xiao ESP32 Sense.
- Upgrade ESP-IDF to v5.3.1.
- Observe that PeerConnection moves to Completed state, but no video is transmitted.

Does this repository support ESP-IDF v5.3.1? If not, is there any plan to add support for it?

Thanks!

sepfy, 2024-09-16:
Hello, i just tested with idf v5.3.1 and it works for me. `PeerConnectionState: 5` means failed. it could be connection issue.

BehicKlncky, 2024-09-16:
@sepfy Thank you! I was able to get it working as well after bunch of trials.

closed 2024-09-16 by BehicKlncky

wilsonlv, 2026-01-05:
@BehicKlncky I encountered the same problem. Could you tell us how you solved it?

BehicKlncky, 2026-01-05:
@wilsonlv 

Hi!

There was a connection issue I believe. I tried another networks at that time and it worked somehow.