sepfy/libpeer #130: Error "Failed to sendto: Not enough space" occurring frequently

BehicKlncky, 2024-09-13:
Hi,

I am encountering the following errors consistently while working with WebRTC on Xiao ESP32 Sense using libpeer:

`ERROR   ./managed_components/sepfy__libpeer/src/socket.c  142  Failed to sendto: Not enough space`
`ERROR   ./managed_components/sepfy__libpeer/src/buffer.c     45       no enough space`

This error appears frequently, and I'm not sure if it's due to network buffer limitations or something else within the library. Is this a known issue, and are there any recommended solutions or workarounds to resolve this?

sepfy, 2024-09-16:
Based on my own experience, it could be an issue with the antenna. You could try adjusting the WiFi antenna.

BehicKlncky, 2024-09-16:
@sepfy Thank you! I’ll try adjusting the antenna as you suggested and see if that helps. I hadn’t considered the possibility that it might be related to Wi-Fi signal strength. Just to clarify, could the weak signal or slow connection cause images or data to build up in the buffer, since the connection isn't fast enough to send the data in real-time? If you have any insights on that or other suggestions, I’d be happy to try them. I'll also share any updates once I’ve tested this.

sepfy, 2024-09-20:
When I use the XIAO, adjusting the antenna can resolve the `Failed to sendto: Not enough space` error. However, if it's `not enough space`, there might be other issues, and I'll look into optimizing the process.

BehicKlncky, 2024-10-07:
Hi @sepfy 

Thanks for the suggestion. I hope that the issue can be resolved with a more optimized method rather than just the antenna adjustment. For now, I'll go ahead and close this issue.

closed 2024-10-07 by BehicKlncky