sepfy/libpeer #133: Connection fails when connecting from different network

karaxuna, 2024-09-18:
I'm trying to connect to esp32-cam from react-native application with https://github.com/react-native-webrtc/react-native-webrtc package. When phone and cam is on the same network, it works fine, but when on different, ice connection state changes to failed. I'm sending offer from react native app when all candidates are gathered:

Offer (from react-native app):

```
v=0
o=- 8521413602419897820 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE 0
a=extmap-allow-mixed
a=msid-semantic: WMS
m=application 15509 UDP/DTLS/SCTP webrtc-datachannel
c=IN IP4 185.115.4.96
a=candidate:4113440344 1 udp 2122260223 172.20.10.2 54848 typ host generation 0 network-id 1 network-cost 10
a=candidate:3050597617 1 udp 1686052607 185.115.4.96 15509 typ srflx raddr 172.20.10.2 rport 54848 generation 0 network-id 1 network-cost 10
a=candidate:193252044 1 tcp 1518280447 172.20.10.2 62123 typ host tcptype passive generation 0 network-id 1 network-cost 10
a=ice-ufrag:VlEl
a=ice-pwd:76oey5NHLeCv7rYwlwJYObcH
a=ice-options:trickle renomination
a=fingerprint:sha-256 7E:31:4D:C4:B2:63:8C:90:1E:48:CD:F8:41:3A:9A:41:15:8C:E1:E9:F2:20:2C:45:D1:98:6D:97:20:B2:1F:54
a=setup:actpass
a=mid:0
a=sctp-port:5000
a=max-message-size:262144
```

Answer (from esp32 cam):

```
v=0
o=- 1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: iot
a=group:BUNDLE 0
m=application 50712 UDP/DTLS/SCTP webrtc-datachannel
a=mid:0
a=sctp-port:5000
c=IN IP4 0.0.0.0
a=max-message-size:262144
a=fingerprint:sha-256 00:C6:DC:62:94:D1:9A:94:A1:DD:EB:27:BC:E0:E8:DB:8F:79:BD:33:E4:E9:A6:E9:49:B7:02:FF:F7:2C:00:1C
a=setup:passive
a=ice-ufrag:6uSh
a=ice-pwd:6uShFC2K0fMkhd4MT2xZRptB
a=candidate:0 1 UDP 2127683327 192.168.100.23 53726 typ host
a=candidate:2 1 UDP 1691475711 94.240.xxx.x 53726 typ srflx raddr 0.0.0.0 rport 0
```

What's wrong with these sdps?

sepfy, 2024-09-20:
Hmm... it looks normal, haha. Are both devices using different routers? or 4g?

karaxuna, 2024-09-23:
Offerer uses 4g and answerer uses home router.