sepfy/libpeer #139: Show example of ESP32 + WHIP in examples

Sean-Der, 2024-09-23:
@sepfy would you be ok if I modified examples/esp32 to allow users to do WHIP with it? Make make it something users can configure in `idf.py menuconfig` ?

Would you also mind tagging a new release? I am going to write a blogpost of how to use WHIP + libpeer against Pion and v0.0.2 doesn't have the fixes to work with Pion.

thank you!

sepfy, 2024-09-23:
Of course. Let me know which commit you'd like to use as the first release for your blog post.

Sean-Der, 2024-09-23:
Should I make `examples/esp32-whip` or add WHIP options to `examples/esp32`?

I think adding WHIP to `examples/esp32` is better. I want to implement it however you think is best though!

sepfy, 2024-09-23:
yes. adding to `examples/esp32` is great. Thanks!