sepfy/libpeer #140: Fix (or disable) `-Wincompatible-pointer-types`

Sean-Der, 2024-09-23:
```
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c: In function 'peer_signaling_http_post':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c:259:17: error: assignment to 'TransportRecv_t' {aka 'long int (*)(struct NetworkContext *, void *, unsigned int)'} from incompatible pointer type 'int (*)(NetworkContext_t *, void *, size_t)' {aka 'int (*)(struct NetworkContext *, void *, unsigned int)'} [-Wincompatible-pointer-types]
  259 |   trans_if.recv = ssl_transport_recv;
      |                 ^
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c:260:17: error: assignment to 'TransportSend_t' {aka 'long int (*)(struct NetworkContext *, const void *, unsigned int)'} from incompatible pointer type 'int (*)(NetworkContext_t *, const void *, size_t)' {aka 'int (*)(struct NetworkContext *, const void *, unsigned int)'} [-Wincompatible-pointer-types]
  260 |   trans_if.send = ssl_transport_send;
      |                 ^
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c: In function 'peer_signaling_mqtt_connect':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c:332:23: error: assignment to 'TransportRecv_t' {aka 'long int (*)(struct NetworkContext *, void *, unsigned int)'} from incompatible pointer type 'int (*)(NetworkContext_t *, void *, size_t)' {aka 'int (*)(struct NetworkContext *, void *, unsigned int)'} [-Wincompatible-pointer-types]
  332 |   g_ps.transport.recv = ssl_transport_recv;
      |                       ^
/project/deps/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_signaling.c:333:23: error: assignment to 'TransportSend_t' {aka 'long int (*)(struct NetworkContext *, const void *, unsigned int)'} from incompatible pointer type 'int (*)(NetworkContext_t *, const void *, size_t)' {aka 'int (*)(struct NetworkContext *, const void *, unsigned int)'} [-Wincompatible-pointer-types]
  333 |   g_ps.transport.send = ssl_transport_send;
      |                       ^

```

I also gets lots of them from libsrtp

```
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c: In function 'srtp_cipher_output':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:108:56: warning: passing argument 3 of 'c->type->encrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  108 |     return (((c)->type)->encrypt(((c)->state), buffer, num_octets_to_output));
      |                                                        ^~~~~~~~~~~~~~~~~~~~
      |                                                        |
      |                                                        uint32_t * {aka long unsigned int *}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:108:56: note: expected 'unsigned int *' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c: In function 'srtp_cipher_encrypt':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:119:56: warning: passing argument 3 of 'c->type->encrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  119 |     return (((c)->type)->encrypt(((c)->state), buffer, num_octets_to_output));
      |                                                        ^~~~~~~~~~~~~~~~~~~~
      |                                                        |
      |                                                        uint32_t * {aka long unsigned int *}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:119:56: note: expected 'unsigned int *' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c: In function 'srtp_cipher_decrypt':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:130:56: warning: passing argument 3 of 'c->type->decrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  130 |     return (((c)->type)->decrypt(((c)->state), buffer, num_octets_to_output));
      |                                                        ^~~~~~~~~~~~~~~~~~~~
      |                                                        |
      |                                                        uint32_t * {aka long unsigned int *}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:130:56: note: expected 'unsigned int *' but argument is of type 'uint32_t *' {aka 'long unsigned int *'}
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c: In function 'srtp_cipher_type_test':
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:297:49: warning: passing argument 3 of 'srtp_cipher_encrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  297 |         status = srtp_cipher_encrypt(c, buffer, &len);
      |                                                 ^~~~
      |                                                 |
      |                                                 unsigned int *
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:113:49: note: expected 'uint32_t *' {aka 'long unsigned int *'} but argument is of type 'unsigned int *'
  113 |                                       uint32_t *num_octets_to_output)
      |                                       ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:398:49: warning: passing argument 3 of 'srtp_cipher_decrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  398 |         status = srtp_cipher_decrypt(c, buffer, &len);
      |                                                 ^~~~
      |                                                 |
      |                                                 unsigned int *
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:124:49: note: expected 'uint32_t *' {aka 'long unsigned int *'} but argument is of type 'unsigned int *'
  124 |                                       uint32_t *num_octets_to_output)
      |                                       ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/project/deps/libpeer/examples/esp32/managed_components/sepfy__srtp/libsrtp/crypto/cipher/cipher.c:520:49: warning: passing argument 3 of 'srtp_cipher_encrypt' from incompatible pointer type [-Wincompatible-pointer-types]
  520 |         status = srtp_cipher_encrypt(c, buffer, &length);
      |                                                 ^~~~~~~
      |                                                 |
      |                                                 unsigned int *
```

Sean-Der, 2024-09-23:
To fix them quickly I did

```
diff --git a/examples/esp32/main/CMakeLists.txt b/examples/esp32/main/CMakeLists.txt
index 424fc44..b622c63 100644
--- a/examples/esp32/main/CMakeLists.txt
+++ b/examples/esp32/main/CMakeLists.txt
@@ -4,3 +4,9 @@ idf_component_register(SRCS
 )

 target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")
+
+idf_component_get_property(lib sepfy__srtp COMPONENT_LIB)
+target_compile_options(${lib} PRIVATE -Wno-error=incompatible-pointer-types)
+
+idf_component_get_property(lib sepfy__libpeer COMPONENT_LIB)
+target_compile_options(${lib} PRIVATE -Wno-error=incompatible-pointer-types)
```

sepfy, 2024-09-23:
Got it. i'll fix the incompatible pointer types in libpeer, and libsrtp we can use `-Wno-error=incompatible-pointer-types`

hahashiwp, 2025-04-04:
[image]
[image]
[image]我也遇到了这样的错误，还是没有解决啊