sepfy/libpeer #145: Build failed when in o2 optimize level:  'strcat' accessing xxx or more bytes at offsets xxx and xxx may overlap 1 byte at offset xxx

qingwa2009, 2024-09-28:
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c       
In function 'peer_connection_state_new',
    inlined from 'peer_connection_loop' at D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpe                     eer/src/peer_connection.c:367:9:
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c:309:5: error:                          'strcat' accessing 20497 or more bytes at offsets 10780 and 31276 may overlap 1 byte at offset 31276 [                            -Werror=restrict]
  309 |     strcat(pc->local_sdp.content, description);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c:320:7: error:                          'strcat' accessing 20497 or more bytes at offsets 10780 and 31276 may overlap 1 byte at offset 31276 [                            -Werror=restrict]
  320 |       strcat(pc->local_sdp.content, description);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c:328:7: error:                          'strcat' accessing 20497 or more bytes at offsets 10780 and 31276 may overlap 1 byte at offset 31276 [                            -Werror=restrict]
  328 |       strcat(pc->local_sdp.content, description);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c:335:7: error:                          'strcat' accessing 20497 or more bytes at offsets 10780 and 31276 may overlap 1 byte at offset 31276 [                            -Werror=restrict]
  335 |       strcat(pc->local_sdp.content, description);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/desktop/libpeer/examples/esp32/managed_components/sepfy__libpeer/src/peer_connection.c:345:5: error:                          'strcat' accessing 20497 or more bytes at offsets 10780 and 31276 may overlap 1 byte at offset 31276 [                            -Werror=restrict]
  345 |     strcat(pc->local_sdp.content, description);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors

sepfy, 2024-10-10:
Can you provide more details about your compilation environment? Includes IDF version. Because I haven't encountered this problem.

qingwa2009, 2024-10-10:
I just set the optimization level to -O2, it build failed on window11 or linux on wsl2
i use idf 5.2.2
[cid:28f6a9bd-bcc2-4e86-adc5-ea2ccf3fbff2]
[cid:9074a57e-8c89-4442-842a-264d462017cf]

commit 20c73ee2 (2025-04-05): Replace strcat with snprintf to prevent buffer overflow (Fixes #145)
closed 2025-04-05 by sepfy