sepfy/libpeer #146: Two way audio communication from web to device.

Param-21, 2024-10-02:
Hi, i have taken the generic example and it is working perfectly i,e sending audio and video from the wav file to the web, Now i have open  web in different pc / device (device) and i want to listen audio from the web running on different device into my device's speaker where i will be running the c code. Could you please help me with this?.

sepfy, 2024-10-10:
Hi, you can add a callback function to receive the audio data from the web
https://github.com/sepfy/libpeer/blob/main/src/peer_connection.h#L65

Param-21, 2025-01-08:
Hey @finch71 , Thanks for the reply, I have now successfully implemented the real-time audio streaming on the webrtc server by modifying the code and with the help of ALSA libs, i am able to send the audio to the webrtc server without any delay or distortion. 
I have implemented on the generic example (linux) itself...