sepfy/libpeer #158: connect with custom server

Param-21, 2024-10-17:
Hi,

I want to connect with custom server instead of "https://sepfy.github.io/webrtc/?deviceId=1". Now i have changed SERVER_CONFIG in peer_signaling.h and mqtt connection establish. now when they shared offer data on that time i'm getting some errors. So is there any example code available of server side which is working with /example/generic.

Thanks

sepfy, 2024-10-18:
@Param-21 Hello, what's your error? What do you mean by your server example? If you are referring to frontend code, you can refer to this repo.
https://github.com/sepfy/sepfy.github.io/blob/main/webrtc/index.html

sepfy, 2024-10-18:
@HaARDIKSD297 not support yet

HaARDIKSD297, 2024-10-25:
@sepfy Is there any available code that takes audio from server and plays on client's speaker, Currently I am using generic example code which is sending audio/video from client to server.

Thanks,

closed 2024-12-30 by Param-21