sepfy/libpeer #162: The critical handling of circular buffer overflow carries the risk of memory misalignment and access exceptions.

zelzmz, 2024-10-30:
Hi，
In buffer. c, the handling of overflow situations by the buffer_push_tail function is as follows:

```
if (tail_end < rb->tail) {

    if (rb->head < align_size) {

      LOGE("no enough space");
      return -1;
    }

    int *p = (int*)(rb->data + rb->tail);
    *p = size;
    memcpy(rb->data, data, size);    
    rb->tail = size;

  }
```

Size is the true length of the data and is not aligned. Storing it in strictly aligned memory can cause CPU errors.
My modification is like this:

```
int buffer_push_tail(Buffer* rb, const uint8_t* data, int size) {
  int free_space = (rb->size + rb->head - rb->tail - 1) % rb->size;

  int align_size = ALIGN32(size + 4);

  if (align_size > free_space) {
    LOGE("no enough space");
    return -1;
  }

  int tail_end = (rb->tail + align_size) % rb->size;

  if (tail_end < rb->tail) {

    if (rb->head < align_size) {

      LOGE("no enough space");
      return -1;
    }

    **int* p = (int*)rb->data;**
    *p = size;
    **memcpy(rb->data + 4, data, size);**
    **rb->tail = align_size;**

  } else {
    int* p = (int*)(rb->data + rb->tail);
    *p = size;
    memcpy(rb->data + rb->tail + 4, data, size);
    rb->tail = tail_end;
  }

  return size;
}

uint8_t* buffer_peak_head(Buffer* rb, int* size) {
  if (!rb || rb->head == rb->tail) {

    return NULL;
  }

  *size = *((int*)(rb->data + rb->head));

  int align_size = ALIGN32(*size + 4);

  int head_end = (rb->head + align_size) % rb->size;

  if (head_end < rb->head) {

    **return rb->data + 4;**

  } else {

    return rb->data + (rb->head + 4);
  }
}

void buffer_pop_head(Buffer* rb) {
  if (!rb || rb->head == rb->tail) {
    return;
  }

  int* size = (int*)(rb->data + rb->head);

  int align_size = ALIGN32(*size + 4);

  int head_end = (rb->head + align_size) % rb->size;

  if (head_end < rb->head) {

    **rb->head = 0;**

  } else {

    rb->head = rb->head + align_size;
  }
}
```

When the last position of the circular buffer is not enough to store the next data, if there is still space at the front, align the length (4)+data and place it at the front.
This modification is relatively simple but wastes space. Let's see if we need to further optimize it.

zelzmz, 2024-10-30:
My modification is like this:

```
int buffer_push_tail(Buffer* rb, const uint8_t* data, int size) {
  int free_space = (rb->size + rb->head - rb->tail - 1) % rb->size;

  int align_size = ALIGN32(size) + 4;

  if (align_size > free_space) {
    LOGE("no enough space");
    return -1;
  }

  int tail_end = (rb->tail + align_size) % rb->size;

  if (tail_end < rb->tail) {

    if (rb->head < align_size) {
      LOGE("no enough space");
      return -1;
    }

    int* p = (int*)(rb->data + rb->tail);
    *p = size;
    memcpy(rb->data, data, size);
    rb->tail = ALIGN32(size);

  } else {
    int* p = (int*)(rb->data + rb->tail);
    *p = size;
    memcpy(rb->data + rb->tail + 4, data, size);
    rb->tail = tail_end;
  }

  return size;
}

uint8_t* buffer_peak_head(Buffer* rb, int* size) {
  if (!rb || rb->head == rb->tail) {

    return NULL;
  }

  *size = *((int*)(rb->data + rb->head));

  int align_size = ALIGN32(*size) + 4;

  int head_end = (rb->head + align_size) % rb->size;

  if (head_end < rb->head) {
    return rb->data;

  } else {

    return rb->data + (rb->head + 4);
  }
}

void buffer_pop_head(Buffer* rb) {
  if (!rb || rb->head == rb->tail) {
    return;
  }

  int* size = (int*)(rb->data + rb->head);

  int align_size = ALIGN32(*size) + 4;

  int head_end = (rb->head + align_size) % rb->size;

  if (head_end < rb->head) {
    rb->head = ALIGN32(*size);

  } else {

    rb->head = rb->head + align_size;
  }
}
```

This modification is relatively simple but wastes space. Let's see if we need to further optimize it.

zelzmz, 2024-11-18:
> 在上面，如果你使用 'rb->tail = ALIGN32（size）;'，那么你也应该使用 '*p = ALIGN32（size）;'。

Only data has the risk of misalignment, size is fixed (int).

sepfy, 2024-12-30:
@zelzmz Hello, could you give me a example that I can add it to the test? Thanks

lucle-bonboncar, 2025-01-14:
hello, i have encounter this log:
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space
ERROR   /Users/lucthe/libpeer/src/buffer.c      45      no enough space
ERROR   /Users/lucthe/libpeer/src/socket.c      128     Failed to sendto: Not enough space

zelzmz, 2025-01-14:
> @zelzmz Hello, could you give me a example that I can add it to the test? Thanks

This is related to the chip. My SRAM requires strict 4-byte alignment, and if it is not aligned, the CPU will throw an unaligned exception.
Before modification, every time the ring buffer is exactly full, there will be a problem because it involves tail handling of the ring buffer.

zelzmz, 2025-01-14:
> hello, i have encounter this log: ERROR /Users/lucthe/libpeer/src/buffer.c 45 no enough space ERROR /Users/lucthe/libpeer/src/socket.c 128 Failed to sendto: Not enough space ERROR [...]

Due to the fact that data transmission and reception in libpeer are handled within the internal peer-connection-loop task, the efficiency is not very good, resulting in congestion. My solution is to create another task specifically for sending.

sepfy, 2025-06-08:
Removed buffer/queue in library