sepfy/libpeer #163: Client stream media from webrtc server mbedtls_ssl_handshake returned -0x2700

ngvanhak49, 2024-11-16:
Dear,

I have setup a whep player to webrtc server on ESP, so I have a problem with connect media stream after get offer successs.
Could you give me advise? Thanks!
More log for details:
```
Response Status: 201
Response Body: v=0
o=SRS/5.0.213(Bee) 140028756298976 2 IN IP4 0.0.0.0
s=SRSPlaySession
t=0 0
a=ice-lite
a=group:BUNDLE audio
a=msid-semantic: WMS live/livestream1234567
m=audio 9 UDP/TLS/RTP/SAVPF 63 111
c=IN IP4 0.0.0.0
a=ice-ufrag:37121s86
a=ice-pwd:1269fo09450td9a28dlw0su5h34kz166
a=fingerprint:sha-256 C8:E3:27:E0:7F:97:24:4E:F9:A3:0F:B0:20:3F:FC:10:DF:F6:7E:2E:45:1D:53:4D:20:D6:5D:92:76:0F:27:99
a=setup:passive
a=mid:audio
a=sendonly
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:63 red/48000/2
a=rtpmap:111 opus/48000/2
a=fmtp:111 minptime=10;useinbandfec=1
a=ssrc:10177 cname:pp32o77i4o83nc76
a=ssrc:10177 msid:- c6dda156-98d0-4931-99b9-a967ef1821de
a=ssrc:10177 mslabel:-
a=ssrc:10177 label:c6dda156-98d0-4931-99b9-a967ef1821de
a=candidate:0 1 udp 2130706431 42.96.40.173 8000 typ host generation 0

INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/peer_connection.c      490      peer_connection_set_remote_description

INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/peer_connection.c      525      update pc in set remote description

INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/peer_connection.c      526      fingerprint: C8:E3:27:E0:7F:97:24:4E:F9:A3:0F:B0:20:3F:FC:10:DF:F6:7E:2E:45:1D:53:4D:20:D6:5D:92:76:0F:27:99

I (28953) webrtc: PeerConnectionState: 2
I (28963) webrtc: PeerConnectionState: 3
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x2700

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7700

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7700

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7700

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     407    failed! mbedtls_ssl_handshake returned -0x7280

ERROR   D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     439    no remote fingerprint
INFO    D:/Works/AMZ/devices/audio_ip/esp32/libpeer/src/dtls_srtp.c     419    dtls_srtp role: 0
```

sepfy, 2024-12-31:
@ngvanhak49 hello, could you try this version: fcb700062c4846a1693ebf77f53f52a13ae363ad ?

ngvanhak49, 2024-12-31:
Thanks sepfy, weiph1029,

This issue fixed by updated mbedtls .

closed 2024-12-31 by ngvanhak49