sepfy/libpeer #167: binding request timeout

laumy007, 2024-12-06:
两个设备属于对称型穿透，采集端的设备使用stun链接，接收端的设备使用turn链接，通过turn服务帮忙转发，发现测试一段时间后流就异常停止了，设备端打印“binding request timeout”，状态切换为closed，问题是必现的，不知是否有什么排查方法。

DEBUG   libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG   libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG   libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG   libpeer/src/peer_connection.c        119     RTCP_PSFB 15
DEBUG   libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG   libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG   libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG    libpeer/src/peer_connection.c        119     RTCP_PSFB 15
DEBUG    libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG   libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG    libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG    libpeer/src/peer_connection.c        119     RTCP_PSFB 15
DEBUG   libpeer/src/peer_connection.c        426     agent_recv 61
DEBUG   libpeer/src/dtls_srtp.c      485     DTLS content type: 23
DEBUG   libpeer/src/peer_connection.c        435     Got DTLS data 32
DEBUG   libpeer/src/sctp.c   451     Data of length 4 received on stream 0 with SSN 39, TSN 703256564, PPID 51
DEBUG   libpeer/src/sctp.c   401     Got message (size = 4)
on message: 0 ping, send pong
DEBUG   libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG   libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG   libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG   libpeer/src/peer_connection.c        119     RTCP_PSFB 15
DEBUG   libpeer/src/peer_connection.c        426     agent_recv 57
DEBUG   libpeer/src/dtls_srtp.c      485     DTLS content type: 23
DEBUG   libpeer/src/peer_connection.c        435     Got DTLS data 28
DEBUG   libpeer/src/sctp.c   451     Data of length 12 received on stream 0 with SSN 0, TSN 0, PPID 0
DEBUG   libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG    libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG    libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG    libpeer/src/peer_connection.c        119     RTCP_PSFB 15
DEBUG    libpeer/src/peer_connection.c        426     agent_recv 70
DEBUG    libpeer/src/peer_connection.c        429     Got RTCP packet
DEBUG    libpeer/src/peer_connection.c        103     RTCP_PR
DEBUG    libpeer/src/peer_connection.c        119     RTCP_PSFB 15
INFO        libpeer/src/peer_connection.c        459     binding request timeout
state is changed: closed

laumy007, 2024-12-09:
来信已经收到！

Sean-Der, 2024-12-30:
I think we need to update libpeer to take SCTP/RTP/RTCP traffic into account. 

A remote peer doesn't need to send STUN Bindings if it has sent other traffic.

laumy007, 2024-12-30:
来信已经收到！

amr-kasem, 2025-02-26:
#205 check this PR I've added a periodic stun heartbeat

laumy007, 2025-02-26:
来信已经收到！

bailiqun, 2025-03-11:
any solution above that? same problem

ankit-chotaliya, 2025-06-05:
I'm facing similar issue. any solution or quick tip around this?

laumy007, 2025-06-05:
来信已经收到！

sepfy, 2025-06-05:
hello, you can add this to config.h

`#define CONFIG_KEEPALIVE_TIMEOUT 0`

Func86, 2025-12-11:
This can also happen for normal WebRTC connections from browsers, as the STUN bindings should be sent no less than every 15 seconds per [RFC 8445](https://datatracker.ietf.org/doc/rfc8445/), which is longer than the default 10 seconds for `CONFIG_KEEPALIVE_TIMEOUT`.

laumy0929, 2025-12-11:
来信已经收到！