sepfy/libpeer #171: ESP32 Arduino Library

kakopappa, 2024-12-20:
Hey @sepfy 

Great work !

Any plans for an Arduino ESP32 library for small folks like me?

Thanks

HasiEraPro, 2024-12-26:
Hello @sepfy , can you release a arduino esp32 library to stream video p2p

sepfy, 2024-12-30:
Sorry, No such plans in the near future

kakopappa, 2025-01-02:
Hi @sepfy 

I am trying to put together an Arduino Library for ESP32 during the holiday break. This is what I have done so far

1. Make libpeer.so based on code in master
2. Make an Arduino library based on libpeer.so and header files. 
3. Flash the sample [sketch](https://github.com/kakopappa/esp32-arduino-webrtc/blob/main/examples/libpeer-test/libpeer-test.ino), it connects to the MQTT server and wait
3. When I initiate a connect from https://sepfy.github.io/libpeer/, I can see the 

`oniceconnectionstatechange` gets called with state = 2 and then state = 3 and crashes. When I decode the error it looks like this

```
Decoding stack results
0x40119b86: mbedtls_ssl_handshake_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library\ssl_tls.c:4563
0x40119bcc: mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library\ssl_tls.c:4610
0x4016ec9d: dtls_srtp_do_handshake at /Users/aruna/Documents/projects/libpeer/src\dtls_srtp.c:369
0x4016ecea: dtls_srtp_handshake_server at /Users/aruna/Documents/projects/libpeer/src\dtls_srtp.c:386
0x4016f005: dtls_srtp_handshake at /Users/aruna/Documents/projects/libpeer/src\dtls_srtp.c:424
0x40164ebf: peer_connection_loop at /Users/aruna/Documents/projects/libpeer/src\peer_connection.c:384
0x400d2902: peer_connection_task(void*) at C:\Users\user\Documents\Arduino\libraries\esp32-arduino-webrtc\examples\libpeer-test\libpeer-test.ino:53
0x4008e3d2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:139
```

Issue seems to be coming from mbedtls_ssl_handshake_step.  Have you experience this before or do you have any suggestions to how to fix this? 

I am using Arduino Release v3.1.0 based on ESP-IDF v5.3.2