sepfy/libpeer #194: Unable to connect to the  server from Chrome and Firefox

Param-21, 2025-02-05:
Hi,
After cloning this repository and running the project, I am unable to connect to the server from both Chrome and Firefox browsers. I am attaching screenshots of the issue for better understanding.

FireFox:

[image]

Chrome:

[image]

please update me on this issue as soon as possible, it would be a big help and also it would solve the general issue with server connection.
I hope you do the needful and reply on this issue asap. 
Thanks in advance :)

linuxsky, 2025-03-07:
is this issue fixed ?

Param-21, 2025-03-12:
hi @linuxsky yes, @sepfy  has solved this issue, but i haven't checked this on firefox yet, so not sure about firefox but on chrome this issue has been resolved

sepfy, 2025-03-12:
yes, i've fix this issue, it's just that my MQTT broken is down

closed 2025-03-12 by sepfy