sepfy/libpeer #200: CONFIG_DATA_BUFFER_SIZE

amr-kasem, 2025-02-14:
is there any way to decrease data buffer size to approximately  50KB ?
I did this with this modification 
#ifdef CONFIG_DATA_BUFFER_SIZE
#undef CONFIG_DATA_BUFFER_SIZE
#endif
#define CONFIG_DATA_BUFFER_SIZE (SCTP_MTU * 48)

but I got this issue 
I (14091) example_common: Connected to example_netif_sta
I (14091) example_common: - IPv4 address: 192.168.1.19,
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_connection.c      165     peer_connection_create

INFO    /home/amr/my-apps/oip/libs/libpeer/src/agent.c32      create IPv4 UDP socket: 54
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_connection.c      179     Datachannel allocates heap size: 57600, 1200
INFO    /home/amr/my-apps/oip/libs/libpeer/src/buffer.c       18      Available heap: 77824 bytes
INFO    /home/amr/my-apps/oip/libs/libpeer/src/buffer.c       28      buffer_new data: 0x3ffec8e0
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_signaling.c       135     Host: 192.168.1.17, Port: 8883, Path: /caller
INFO    /home/amr/my-apps/oip/libs/libpeer/src/ports.c140     Resolved 192.168.1.17 -> 192.168.1.17
INFO    /home/amr/my-apps/oip/libs/libpeer/src/socket.c       230     Connecting to server: 192.168.1.17:8883
INFO    /home/amr/my-apps/oip/libs/libpeer/src/socket.c       236     Server is connected
INFO    /home/amr/my-apps/oip/libs/libpeer/src/ssl_transport.c        106     start to handshake
INFO    /home/amr/my-apps/oip/libs/libpeer/src/ssl_transport.c        114     handshake success
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_signaling.c       449     MQTT_Connect succeeded.
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_signaling.c       465     Subscribing topic /caller/invoke
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_signaling.c       391     MQTT Subscribe succeeded.
I (15741) webrtc: [APP] Free memory: 29276 bytes
I (15741) webrtc: peer_connection_task started
I (15741) webrtc: ping_task started
============= Configuration =============
 URL   : mqtts://192.168.1.17/caller
 Token : 
=========================================
I (43371) webrtc: PeerConnectionState: new
INFO    /home/amr/my-apps/oip/libs/libpeer/src/peer_connection.c      350     ice server: stun:stun.l.google.com:19302
INFO    /home/amr/my-apps/oip/libs/libpeer/src/ports.c140     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/amr/my-apps/oip/libs/libpeer/src/agent.c269     Resolved stun/turn server 74.125.250.129:19302
INFO    /home/amr/my-apps/oip/libs/libpeer/src/agent.c89      ping request success
ERROR   /home/amr/my-apps/oip/libs/libpeer/src/socket.c       256     Failed to send: Connection already in progress
ERROR   /home/amr/my-apps/oip/libs/libpeer/src/ssl_transport.c        141     ssl write error: -0x1
ERROR   /home/amr/my-apps/oip/libs/libpeer/src/socket.c       256     Failed to send: Connection already in progress
ERROR   /home/amr/my-apps/oip/libs/libpeer/src/ssl_transport.c        141     ssl write error: -0x1
ERROR   /home/amr/my-apps/oip/libs/libpeer/src/socket.c       256     Failed to send: Connection already in progress 

any ideas?

sepfy, 2025-02-20:
hello, have you change any flow? Because this problem does not seem to be affected by data size, if you want the decrease data buffer, you only need to modify CONFIG_DATA_BUFFER_SIZE

sepfy, 2025-04-05:
if still has issue, please reopen

closed 2025-04-05 by sepfy