sepfy/libpeer #203: Failed to connect to server libpeer.com(5.78.82.24)

jiankekejian, 2025-02-21:
[root@cenos_1 generic]# ./sample -u mqtts://libpeer.com/public/striped-brave-lion
=========== Parsed Arguments ===========
 URL   : mqtts://libpeer.com/public/striped-brave-lion
 Token : 

INFO    /home/jk/libpeer/libpeer/src/agent.c    32      create IPv4 UDP socket: 3
INFO    /home/jk/libpeer/libpeer/src/peer_connection.c  175     Datachannel allocates heap size: 153600
INFO    /home/jk/libpeer/libpeer/src/peer_connection.c  182     Audio allocates heap size: 332800
INFO    /home/jk/libpeer/libpeer/src/peer_connection.c  195     Video allocates heap size: 332800
INFO    /home/jk/libpeer/libpeer/src/peer_signaling.c   134     Host: libpeer.com, Port: 8883, Path: /public/striped-brave-lion
INFO    /home/jk/libpeer/libpeer/src/ports.c    140     Resolved libpeer.com -> 5.78.82.24
INFO    /home/jk/libpeer/libpeer/src/socket.c   230     **Connecting to server: 5.78.82.24:8883**
ERROR   /home/jk/libpeer/libpeer/src/socket.c   232     **Failed to connect to server**
ERROR   /home/jk/libpeer/libpeer/src/peer_signaling.c   401     ssl transport connect failed
^CERROR /home/jk/libpeer/libpeer/src/peer_signaling.c   462     MQTT_Subscribe failed: Status=MQTTBadParameter.
INFO    /home/jk/libpeer/libpeer/src/peer_signaling.c   548     Disconnected
[root@cenos_1 generic]# 
[root@cenos_1 generic]# 
[root@cenos_1 generic]# **telnet 5.78.82.24 8883
Trying 5.78.82.24...
telnet: connect to address 5.78.82.24: Connection refused**

blaoke, 2025-02-23:
same problem
```sh
$ ./build/examples/generic/sample -u mqtts://libpeer.com/public/spotted-smart-panda
=========== Parsed Arguments ===========
 URL   : mqtts://libpeer.com/public/spotted-smart-panda
 Token : 
========================================
INFO	/home/bu/wendang/code/libpeer/src/agent.c	32	create IPv4 UDP socket: 3
INFO	/home/bu/wendang/code/libpeer/src/peer_connection.c	175	Datachannel allocates heap size: 153600
INFO	/home/bu/wendang/code/libpeer/src/peer_connection.c	182	Audio allocates heap size: 332800
INFO	/home/bu/wendang/code/libpeer/src/peer_connection.c	195	Video allocates heap size: 332800
INFO	/home/bu/wendang/code/libpeer/src/peer_signaling.c	134	Host: libpeer.com, Port: 8883, Path: /public/spotted-smart-panda
INFO	/home/bu/wendang/code/libpeer/src/ports.c	140	Resolved libpeer.com -> 5.78.82.24
INFO	/home/bu/wendang/code/libpeer/src/socket.c	230	Connecting to server: 5.78.82.24:8883
ERROR	/home/bu/wendang/code/libpeer/src/socket.c	232	Failed to connect to server
ERROR	/home/bu/wendang/code/libpeer/src/peer_signaling.c	401	ssl transport connect `failed`
```

benoitclem-ath, 2025-02-25:
Pretty much the same here after a successful build. Any progress or hint on this issue? 
I'll be very keen on helping on this! 
Cheers
Clem

```=========== Parsed Arguments ===========
 URL   : mqtts://libpeer.com/public/spotted-kind-panda
 Token : 
========================================
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/agent.c 32      create IPv4 UDP socket: 3
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/peer_connection.c       175     Datachannel allocates heap size: 153600
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/peer_connection.c       182     Audio allocates heap size: 332800
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/peer_connection.c       195     Video allocates heap size: 332800
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/peer_signaling.c        134     Host: libpeer.com, Port: 8883, Path: /public/spotted-kind-panda
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/ports.c 140     Resolved libpeer.com -> 5.78.82.24
INFO    /home/clement/work/wr/visionaryDuoS/libpeer/src/socket.c        230     Connecting to server: 5.78.82.24:8883
ERROR   /home/clement/work/wr/visionaryDuoS/libpeer/src/socket.c        232     Failed to connect to server
ERROR   /home/clement/work/wr/visionaryDuoS/libpeer/src/peer_signaling.c        401     ssl transport connect failed
```

sepfy, 2025-02-26:
hey. i've solved this issue.

blaoke, 2025-02-28:
@sepfy  How did you solve the problem，Anyone can help?