sepfy/libpeer #207: ZLmediakit的示例中https://github.com/sepfy/libpeer/wiki/WebRTC-Server的参考代码和最新版本已经不匹配了

wzh469649067, 2025-03-10:


wzh469649067, 2025-03-11:
wzh@wzh:~/work/test_old_libpeer/libpeer/build$ ./examples/generic/sample 
open https://sepfy.github.io/webrtc?deviceId=(null)
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/agent.c	32	create IPv4 UDP socket: 3
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_connection.c	176	Datachannel allocates heap size: 153600
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_connection.c	183	Audio allocates heap size: 332800
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_connection.c	196	Video allocates heap size: 332800
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_signaling.c	500	HTTP Host: 47.97.49.179, Port: 443, Path: /index/api/webrtc?app=live&stream=test&type=push
state is changed: new
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_connection.c	297	ice server: stun:stun.l.google.com:19302
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/ports.c	140	Resolved stun.l.google.com -> 74.125.250.129
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/agent.c	269	Resolved stun/turn server 74.125.250.129:19302
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/ports.c	140	Resolved 47.97.49.179 -> 47.97.49.179
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/socket.c	230	Connecting to server: 47.97.49.179:443
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/socket.c	236	Server is connected
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/ssl_transport.c	107	start to handshake
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/ssl_transport.c	115	handshake success
INFO	/home/wzh/work/test_old_libpeer/libpeer/src/peer_signaling.c	279	Received HTTP response from 47.97.49.179/index/api/webrtc?app=live&stream=test&type=push
Response Headers: Connection: keep-alive
Content-Length: 1854
Content-Type: application/json; charset=utf-8
Date: Tue, Mar 11 2025 05:43:34 GMT
Keep-Alive: timeout=30, max=100
Server: ZLMediaKit(git hash:83afaff/2025-03-09T12:11:58+08:00,branch:master,build time:2025-03-11T13:07:33)

{
	"code" : 0,
	"id" : "rBB4tB9AH0A=_1",
	"sdp" : "v=0\r\no=- 1495799811084970 1495799811084970 IN IP4 47.97.49.179\r\ns=-\r\nt=0 0\r\na=group:BUNDLE video audio datachannel\r\na=msid-semantic: iot\r\na=ice-lite\r\nm=video 8000 UDP/TLS/RTP/SAVPF 96\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:video\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:96 H264/90000\r\na=fmtp:96 level-asymmetry-allowed=1;profile-level-id=42e01f\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=audio 8000 UDP/TLS/RTP/SAVP 8\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:audio\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:8 PCMA/8000/1\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=application 8000 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:datachannel\r\na=ice-lite\r\na=sctp-port:5000\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidwebrtc/peer/jsonrpc
Response Status: 200
Response Body: {
	"code" : 0,
	"id" : "rBB4tB9AH0A=_1",
	"sdp" : "v=0\r\no=- 1495799811084970 1495799811084970 IN IP4 47.97.49.179\r\ns=-\r\nt=0 0\r\na=group:BUNDLE video audio datachannel\r\na=msid-semantic: iot\r\na=ice-lite\r\nm=video 8000 UDP/TLS/RTP/SAVPF 96\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:video\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:96 H264/90000\r\na=fmtp:96 level-asymmetry-allowed=1;profile-level-id=42e01f\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=audio 8000 UDP/TLS/RTP/SAVP 8\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:audio\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:8 PCMA/8000/1\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=application 8000 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:datachannel\r\na=ice-lite\r\na=sctp-port:5000\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidwebrtc/peer/jsonrpc

之后就没有后续了

wzh469649067, 2025-03-11:
2025-03-11 13:43:34.846 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:415 CreateSslCtx | setting SRTP cryptoSuites for DTLS: SRTP_AEAD_AES_256_GCM:SRTP_AEAD_AES_128_GCM:SRTP_AES128_CM_SHA1_80:SRTP_AES128_CM_SHA1_32
2025-03-11 13:43:34.846 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:497 GenerateFingerprints | sha-1   fingerprint: 42:DF:6C:8F:C4:17:86:28:E0:94:E9:96:DE:77:1D:92:F6:AB:99:D0
2025-03-11 13:43:34.846 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:497 GenerateFingerprints | sha-224 fingerprint: FA:C0:5C:5F:E4:D2:71:8D:68:9D:F8:54:78:D5:90:8B:74:73:B3:F5:59:87:5C:7E:69:D0:11:1F
2025-03-11 13:43:34.846 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:497 GenerateFingerprints | sha-256 fingerprint: 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB
2025-03-11 13:43:34.847 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:497 GenerateFingerprints | sha-384 fingerprint: CB:DA:05:58:6A:13:22:82:49:88:3E:FC:B4:F5:B8:4D:E5:60:9F:66:2A:DD:84:F8:44:27:76:41:74:C4:DF:91:C8:86:78:A1:38:B6:1C:E1:81:75:DE:CD:C5:E6:78:8E
2025-03-11 13:43:34.847 D [MediaServer] [42565-event poller 0] DtlsTransport.cpp:497 GenerateFingerprints | sha-512 fingerprint: 20:64:DD:E9:3D:1E:CA:CC:B9:15:4A:99:9D:E0:FA:F5:DC:17:A4:0B:C8:C2:74:9D:DF:76:99:7F:39:4B:CC:C5:12:8F:75:38:2C:DB:FE:0A:55:6B:2D:79:37:10:DB:8D:C9:93:44:A2:06:DE:23:B7:2A:AB:15:6A:DA:FA:E3:13
2025-03-11 13:43:34.847 D [MediaServer] [42565-event poller 0] WebApi.cpp:274 http api debug | 
# request:
POST /index/api/webrtc?app=live&stream=test&type=push
# header:
Connection : keep-alive
Content-Length : 1774
Content-Type : application/sdp
Host : 47.97.49.179
User-Agent : my-platform-name
# content:
v=0
o=- 1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: iot
a=group:BUNDLE video audio datachannel
m=video 9 UDP/TLS/RTP/SAVPF 96 102
a=rtcp-fb:102 nack
a=rtcp-fb:102 nack pli
a=fmtp:96 profile-level-id=42e01f;level-asymmetry-allowed=1
a=fmtp:102 profile-level-id=42e01f;packetization-mode=1;level-asymmetry-allowed=1
a=rtpmap:96 H264/90000
a=rtpmap:102 H264/90000
a=ssrc:1 cname:webrtc-h264
a=sendrecv
a=mid:video
c=IN IP4 0.0.0.0
a=rtcp-mux
a=fingerprint:sha-256 91:DE:CF:1E:83:D5:0B:EF:72:C6:E9:B9:E4:EC:AD:1C:EB:94:C8:64:21:CF:EB:A7:CE:83:41:67:5B:18:16:02
a=setup:passive
a=ice-ufrag:qb3e
a=ice-pwd:qb3e7etQQmPvJ0bDeC3uxITA
a=candidate:0 1 UDP 2129301247 192.168.112.207 60046 typ host
a=candidate:2 1 UDP 1693093631 115.216.52.172 60046 typ srflx raddr 0.0.0.0 rport 0
m=audio 9 UDP/TLS/RTP/SAVP 8
a=rtpmap:8 PCMA/8000
a=ssrc:4 cname:webrtc-pcma
a=sendrecv
a=mid:audio
c=IN IP4 0.0.0.0
a=rtcp-mux
a=fingerprint:sha-256 91:DE:CF:1E:83:D5:0B:EF:72:C6:E9:B9:E4:EC:AD:1C:EB:94:C8:64:21:CF:EB:A7:CE:83:41:67:5B:18:16:02
a=setup:passive
a=ice-ufrag:qb3e
a=ice-pwd:qb3e7etQQmPvJ0bDeC3uxITA
a=candidate:0 1 UDP 2129301247 192.168.112.207 60046 typ host
a=candidate:2 1 UDP 1693093631 115.216.52.172 60046 typ srflx raddr 0.0.0.0 rport 0
m=application 50712 UDP/DTLS/SCTP webrtc-datachannel
a=mid:datachannel
a=sctp-port:5000
c=IN IP4 0.0.0.0
a=max-message-size:262144
a=fingerprint:sha-256 91:DE:CF:1E:83:D5:0B:EF:72:C6:E9:B9:E4:EC:AD:1C:EB:94:C8:64:21:CF:EB:A7:CE:83:41:67:5B:18:16:02
a=setup:passive
a=ice-ufrag:qb3e
a=ice-pwd:qb3e7etQQmPvJ0bDeC3uxITA
a=candidate:0 1 UDP 2129301247 192.168.112.207 60046 typ host
a=candidate:2 1 UDP 1693093631 115.216.52.172 60046 typ srflx raddr 0.0.0.0 rport 0

# response:
{
	"code" : 0,
	"id" : "rBB4tB9AH0A=_1",
	"sdp" : "v=0\r\no=- 1495799811084970 1495799811084970 IN IP4 47.97.49.179\r\ns=-\r\nt=0 0\r\na=group:BUNDLE video audio datachannel\r\na=msid-semantic: iot\r\na=ice-lite\r\nm=video 8000 UDP/TLS/RTP/SAVPF 96\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:video\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:96 H264/90000\r\na=fmtp:96 level-asymmetry-allowed=1;profile-level-id=42e01f\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=audio 8000 UDP/TLS/RTP/SAVP 8\r\nc=IN IP4 47.97.49.179\r\na=rtcp:8000 IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:audio\r\na=ice-lite\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:8 PCMA/8000/1\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\nm=application 8000 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 47.97.49.179\r\na=ice-ufrag:rBB4tB9AH0A=_1\r\na=ice-pwd:usf9Vf8toF5oajV6j4eycw2L\r\na=fingerprint:sha-256 86:FD:E3:7C:9D:F5:5C:FF:35:80:F3:F6:2A:AE:97:EA:CE:18:CA:50:8A:2D:F8:7F:D0:3E:AC:63:2B:85:75:AB\r\na=setup:active\r\na=mid:datachannel\r\na=ice-lite\r\na=sctp-port:5000\r\na=candidate:udpcandidate 1 udp 120 47.97.49.179 8000 typ host\r\na=candidate:tcpcandidate 1 tcp 115 47.97.49.179 8000 typ host tcptype passive\r\n",
	"type" : "answer"
}

平台中的打印是这样的

wzh469649067, 2025-03-11:
已解决，url的问题

closed 2025-03-11 by wzh469649067

xrz1181, 2025-03-25:
大佬，您用的是哪个分支代码？ZLmediakit在虚拟机中部署需要怎么设置的？

lostkk2024, 2025-05-06:
> 已解决，url的问题

我也遇到这个问题了。哥，能指点下是哪个url么。谢谢