sepfy/libpeer #213: Are there plans to support websocket?

wzh469649067, 2025-03-29:
Are there plans to support websocket? The use of websocket for signaling interaction in webrtc applications is a relatively common solution, and libpeer will be more common if it can be supported.

wzh469649067, 2025-04-02:
ERROR	/home/wzh/work/libpeer/src/dtls_srtp.c	412	failed! mbedtls_ssl_handshake returned -0x7280
INFO	/home/wzh/work/libpeer/src/dtls_srtp.c	421	DTLS server handshake done
INFO	/home/wzh/work/libpeer/src/dtls_srtp.c	449	role :[1][(null)]
ERROR	/home/wzh/work/libpeer/src/dtls_srtp.c	462	no remote fingerprint,[MBEDTLS_SSL_IN_CONTENT_LEN:16384]
PEER_CONNECTION_CONNECTED
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:0069: set_timer to 0 ms
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:0069: set_timer to 0 ms
mbedtls_ssl_session_reset ret:0
mbedtls_ssl_set_client_transport_id ret:0
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:0069: set_timer to 0 ms
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_tls.c:3939: => handshake
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:2124: => flush output
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:2133: <= flush output
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_tls12_server.c:4228: server state: 0
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:2124: => flush output
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:2133: <= flush output
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_tls12_server.c:4228: server state: 1
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_tls12_server.c:0914: => parse client hello
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:1926: => fetch input
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:1974: in_left: 0, nb_want: 5
DBG==>/home/wzh/work/libpeer/third_party/mbedtls/library/ssl_msg.c:2011: f_recv_timeout: 0 ms

Try to use wss as signaling server, after SDP exchange, select the candidate, the state of pc->state changes to PEER_CONNECTION_CONNECTED, ssl occurs 0x7280 during connection, check for several days still can not find the method, please ask what method or where the operation is wrong.

sepfy, 2025-04-05:
Can you confirm that your client correctly sets the offer from libpeer? 0x7280 means the client message was not received correctly

leejuen, 2025-04-10:
你可以参考mqtt的逻辑编写websocket的信令服务，这很容易。

wzh469649067, 2025-04-10:
> 你可以参考mqtt的逻辑编写websocket的信令服务，这很容易。

是的，已经实现了

closed 2025-04-10 by wzh469649067