sepfy/libpeer #214: Replacing another test.264 file with the original video used for testing caused the sample to crash

kingoverworld, 2025-03-30:
When I was using other videos for testing, I made a mistake in the sample example under generic. The video fully complies with the H264 raw stream including SPS, PPS and NALU. The information detected using ffprobe is as follows:
ffprobe version N-118926-gd435b641a2 Copyright (c) 2007-2025 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.4.0-1ubuntu1~22.04)
  configuration: --enable-libx264 --enable-demuxer=rtsp --enable-gpl --prefix=/usr
  libavutil      59. 60.100 / 59. 60.100
  libavcodec     61. 33.102 / 61. 33.102
  libavformat    61.  9.107 / 61.  9.107
  libavdevice    61.  4.100 / 61.  4.100
  libavfilter    10.  9.100 / 10.  9.100
  libswscale      8. 13.103 /  8. 13.103
  libswresample   5.  4.100 /  5.  4.100
  libpostproc    58.  4.100 / 58.  4.100
Input #0, h264, from 'test.264':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuv420p(progressive), 384x288 [SAR 213:160 DAR 71:40], 25 fps, 60 tbr, 1200k tbn

When I using sample -u to run webrtc demo, the following messages:
========================================
INFO	/opt/libpeer/src/agent.c	32	create IPv4 UDP socket: 3
INFO	/opt/libpeer/src/peer_connection.c	175	Datachannel allocates heap size: 153600
INFO	/opt/libpeer/src/peer_connection.c	182	Audio allocates heap size: 332800
INFO	/opt/libpeer/src/peer_connection.c	195	Video allocates heap size: 332800
INFO	/opt/libpeer/src/peer_signaling.c	134	Host: libpeer.com, Port: 8883, Path: /public/sleek-smart-fox
INFO	/opt/libpeer/src/ports.c	140	Resolved libpeer.com -> 5.78.82.24
INFO	/opt/libpeer/src/socket.c	230	Connecting to server: 5.78.82.24:8883
INFO	/opt/libpeer/src/socket.c	236	Server is connected
INFO	/opt/libpeer/src/ssl_transport.c	106	start to handshake
INFO	/opt/libpeer/src/ssl_transport.c	114	handshake success
INFO	/opt/libpeer/src/peer_signaling.c	440	MQTT_Connect succeeded.
INFO	/opt/libpeer/src/peer_signaling.c	456	Subscribing topic /public/sleek-smart-fox/invoke
INFO	/opt/libpeer/src/peer_signaling.c	382	MQTT Subscribe succeeded.
state is changed: new
INFO	/opt/libpeer/src/peer_connection.c	345	ice server: stun:stun.l.google.com:19302
INFO	/opt/libpeer/src/ports.c	140	Resolved stun.l.google.com -> 74.125.250.129
INFO	/opt/libpeer/src/agent.c	269	Resolved stun/turn server 74.125.250.129:19302
INFO	/opt/libpeer/src/mdns.c	157	Resolved c35b02c3-c212-4f87-85ab-c6f47928bb7d.local -> 192.168.1.200
INFO	/opt/libpeer/src/mdns.c	85	invalid length
INFO	/opt/libpeer/src/mdns.c	78	not a mDNS response
INFO	/opt/libpeer/src/mdns.c	167	Failed to resolve hostname
WARN	/opt/libpeer/src/ice.c	125	Failed to resolve mDNS address
state is changed: checking
state is changed: connected
INFO	/opt/libpeer/src/dtls_srtp.c	308	Created inbound SRTP session
INFO	/opt/libpeer/src/dtls_srtp.c	328	Created outbound SRTP session
INFO	/opt/libpeer/src/peer_connection.c	436	SCTP create socket
state is changed: completed
DATA_CHANNEL_OPEN: Label=libpeer, sid=0
Segment Fault(core has been dumped)
I carefully checked the video file format and found that it is legal. I don't know why?

kingoverworld, 2025-04-01:
> Hi, I have a few opinions for you:
>
> In reader_h264_find_nalu, you should look for 0x01000000 and 0x010000 as the nalu start byte.
>
> In reader_get_video_frame, you should increase the space for pps_frame and sps_frame.
>
> Fix libpeer example code to suit additional characteristics of your media data
>
> Below is the example for changes:
>
> `
>  uint8_t* reader_h264_find_nalu(uint8_t* buf_start, uint8_t* buf_end) {
>  static const uint32_t nalu_start_4bytecode = 0x01000000;
>  static const uint32_t nalu_start_3bytecode = 0x010000;
>  uint8_t* p = buf_start;
>
> while ((p + 3) < buf_end) {
>  if ((p + 3) < buf_end) {
>  if (memcmp(p, &amp;nalu_start_3bytecode, 3) == 0) {
>  return p;
>  }
>  } else if ((p + 4) < buf_end) {
>  if (memcmp(p, &amp;nalu_start_4bytecode, 4) == 0) {
>  return p;
>  }
> [... 142 lines cut ...]
>
> pstart = pend;
>  return ret;
>  }
>  `

Thank you very much!

sepfy, 2025-04-27:
need `-bf 0`

commit 8e91b307 (2025-04-27): Support different start code in h264 nalu (Fixes #214)
closed 2025-04-27 by sepfy