sepfy/libpeer #215: Issue with TURN Server Binding Request in libpeer

pradhyumnap1999, 2025-04-02:
I have successfully cross-compiled **_libpeer_** for **_QNX_** and am currently testing video streaming between:

- A QNX-based application running libpeer
- A WebRTC-based web client running on macOS

For NAT traversal, I am using **_Coturn_** as my STUN/TURN server.

**Observations**
--> **_STUN Server Works as Expected_**
When using a STUN server in libpeer:

1. ICE candidate collection works correctly.
2. The QNX app generates an SDP offer and shares it with the web client. 
3. After receiving the SDP answer, streaming starts properly. 
4. The binding request is sent and successfully receives a binding response.

--> **_TURN Server Issue: Binding Request Fails_**
When using a TURN server in libpeer:

1.ICE candidate collection works correctly, and relay candidates are obtained.
2. The QNX app generates an SDP offer and shares it with the web client.
3. The web client responds with an SDP answer.

**_Issue_**: After receiving the SDP answer, the binding request is sent, but it fails.

After some time, the connection state changes to "failed".

I have explicitly configured the web client to use only relay candidates for testing. Despite trying various configurations, the issue persists only when using a TURN server.

**Debugging Attempts & Findings**
**Authentication**:
- ICE candidate collection successfully authenticates using the provided username, realm, and credentials in peerconfig.
- However, for the binding request, the username is taken from the remote and local ufrag instead of the explicitly provided credentials.
- The request is marked with short-term credentials in the code.

**Coturn Server Logs**:
- While ICE candidate collection succeeds, the binding request always results in a 401 Unauthorized error.
- This behavior is unexpected since initial authentication is successful, but subsequent binding requests fail.

**Questions**
1. Does libpeer fully support TURN servers for relay candidates, or is this an expected limitation?
2. Has libpeer been tested with specific TURN servers? If so, which ones?
3. Why does the binding request fail with a 401 error despite successful authentication during ICE candidate collection? 
4. Are there any known workarounds for handling TURN authentication issues in libpeer?

Any insights or guidance would be greatly appreciated!

pradhyumnap1999, 2025-04-02:
Thank you for the response. I’d like to clarify the exact problem I’m facing:

When libpeer (QNX app) and the web client (Mac) both use the same TURN server, it does not work.

However, when libpeer uses a STUN server and the web client uses a TURN server, it works fine.

The issue happens only when libpeer is also using TURN.

Issue Recap
ICE candidate collection succeeds in libpeer when using TURN.

The SDP offer is created and shared with the web client, which replies with an SDP answer.

After receiving the SDP answer, the binding request fails, and eventually, the connection state changes to “failed.”

Coturn logs show a 401 Unauthorized error for the binding request, even though initial authentication during ICE candidate collection was successful.

Questions:
Has libpeer been tested when both peers (libpeer and web client) are using the same TURN server?

Is there a known issue where libpeer fails at the binding request stage when using a TURN relay?

Could there be a mismatch in how libpeer constructs TURN authentication for the binding request compared to ICE candidate gathering?

I would appreciate any insights or suggestions on this issue. Thank you!

pradhyumnap1999, 2025-04-03:
Request for More Details on Your Setup

Thanks for looking into this! Since you're not encountering the same issue, I’d like to understand your setup better to see if there are any differences.

Are you using the latest code from the main branch, or have you made any modifications?

How did you set up your TURN server?

Which TURN server are you using?

What configuration options did you set?

How are you handling authentication (e.g., long-term vs. short-term credentials)?

Can you share the exact steps you followed for your setup? That would help me replicate your environment and identify what might be different.

I appreciate your help in troubleshooting this. Thanks!

kingoverworld, 2025-04-23:
INFO	/opt/newdisk/libpeer/src/peer_connection.c	345	ice server: stun:stun.l.google.com:19302
INFO	/opt/newdisk/libpeer/src/ports.c	140	Resolved stun.l.google.com -> 74.125.250.129
INFO	/opt/newdisk/libpeer/src/agent.c	269	Resolved stun/turn server 74.125.250.129:19302
INFO	/opt/newdisk/libpeer/src/mdns.c	157	Resolved 196009dd-8768-479f-8536-714805de327f.local -> 192.168.1.199
INFO	/opt/newdisk/libpeer/src/mdns.c	85	invalid length
INFO	/opt/newdisk/libpeer/src/mdns.c	78	not a mDNS response
INFO	/opt/newdisk/libpeer/src/mdns.c	167	Failed to resolve hostname
WARN	/opt/newdisk/libpeer/src/ice.c	125	Failed to resolve mDNS address