sepfy/libpeer #219: libpeer：no enough space

kingoverworld, 2025-04-13:
I use ffmpeg to decode a mp4 file，and wrapper the raw video packet with PPS, SPS and NALU.  When I used the URL provided by libpeer for testing, I encountered some error messages. The playing screen started normal, but gradually began to lag and could not be played. These errors are as follows:
ERROR	/opt/libpeer/src/buffer.c	53	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space
ERROR	/opt/libpeer/src/buffer.c	45	no enough space

I examine the code and it seems that there is not enough space in the queue, so I suspect that the data provided is too fast, causing the queue to read too slowly and resulting in this situation.。

jcl2023, 2025-05-15:
Encountered the same issue. How to enlarge space size in the source code?

Thanks,

sepfy, 2025-06-08:
Removed buffer/queue in library

closed 2025-06-08 by sepfy