sepfy/libpeer #221: generic sample work, but CPU usage is almost 99%, and  a few frames later crash.

honechris, 2025-04-25:
[root@YEUX /home]# ./sample -u mqtts://libpeer.com/public/spotted-calm-fox
=========== Parsed Arguments ===========
 URL   : mqtts://libpeer.com/public/spotted-calm-fox
 Token :
========================================
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/agent.c  32      create IPv4 UDP socket: 3
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_connection.c        175     Datachannel allocates heap size: 153600
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_connection.c        182     Audio allocates heap size: 332800
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_connection.c        195     Video allocates heap size: 332800
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_signaling.c 134     Host: libpeer.com, Port: 8883, Path: /public/spotted-calm-fox
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/ports.c  140     Resolved libpeer.com -> 5.78.82.24
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/socket.c 230     Connecting to server: 5.78.82.24:8883
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/socket.c 236     Server is connected
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/ssl_transport.c  106     start to handshake
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/ssl_transport.c  114     handshake success
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_signaling.c 440     MQTT_Connect succeeded.
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_signaling.c 456     Subscribing topic /public/spotted-calm-fox/invoke
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_signaling.c 382     MQTT Subscribe succeeded.
state is changed: new
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_connection.c        345     ice server: stun:stun.l.google.com:19302
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/ports.c  140     Resolved stun.l.google.com -> 74.125.250.129
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/agent.c  269     Resolved stun/turn server 74.125.250.129:19302
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/mdns.c   157     Resolved 584b91c2-0c8a-40ca-9cbe-269c5457451d.local -> 192.168.0.14
state is changed: checking
state is changed: connected
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/dtls_srtp.c      308     Created inbound SRTP session
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/dtls_srtp.c      328     Created outbound SRTP session
INFO    /home/chris/Work/Platform_SDKs/p2p/libpeer/src/peer_connection.c        436     SCTP create socket
state is changed: completed
DATA_CHANNEL_OPEN: Label=libpeer, sid=0
on message: 0 ping, send pong
on message: 0 ping, send pong
on message: 0 ping, send pong
on message: 0 ping, send pong
Segmentation fault

honechris, 2025-04-29:
same as https://github.com/sepfy/libpeer/issues/162

sepfy, 2025-05-04:
can you test the latest version? if there is still CPU usage and segv issue, please share your h264 file and test environment